particle-iot/device-os #642: attachInterrupt(..., FALLING) only called on RISING edge

pomplesiegel, 2015-09-25:
Issue could extend beyond this particular case, but this one is confirmed. 

**Setup:** SPST button hooked to input D4 (Debounced w/ an RC, pulled high to 5V with external resistor) and input set to INPUT_PULLDOWN. Input to D4 shown on scope. Left side of image marks transition from 5V -> 0V, when the button is pressed. 

**Issue:** Interrupt clearly occurs on the RISING edge instead of falling EDGE. I can hold down the button for a few seconds and see that the interrupt only occurs when I release it.

[image]

This code successfully replicates the issue:

````cpp
#include "application.h"

SYSTEM_MODE(MANUAL);

const int rotaryPush = D4;

volatile static bool buttonPressed = false;

void pushButtonHandler()
{
  buttonPressed = true; //mark that button was pressed
}

void setup() {
  Serial.begin(9600);
  
  Serial.println("Interrupt test!");

  pinMode(rotaryPush, INPUT_PULLDOWN);

  //Attach interrupt to downfall of pushbutton
  attachInterrupt(rotaryPush, pushButtonHandler, FALLING);
}

void loop() 
{
  if(buttonPressed)
  {
    Serial.println("Button pressed!");
    buttonPressed = false;
  }
}
````

andyw-lala, 2015-09-25:
Does the behaviour change if you configure the pin as INPUT_PULLUP, or just
plain input instead (since you're pulling high externally) ?

I'm not near a core/photon at present, or I'd test myself, sorry.

pomplesiegel, 2015-09-25:
Thanks, @andyw-lala! It appears we still have issues

Just changed the external pull-up resistor to be a pull-down, and changed the input line to
`pinMode(rotaryPush, INPUT_PULLUP);` 
so the switch now moves the signal from 0 -> 5V when depressed, as shown in the new scope shot.

[image]

Here's the strange thing:
I forgot to change the interrupt trigger, so the line still reads
`pinMode(rotaryPush, INPUT_PULLUP);`,
but the interrupt **always** triggered off of the rising edge, and sometimes also triggered off of the falling edge (end of button press). 

Is there something funky going on with the Photon interrupts in general?

pomplesiegel, 2015-09-25:
And even more interestingly, 
`attachInterrupt(rotaryPush, pushButtonHandler, RISING);`
triggers only on the **end** of the above scope shot, as the signal is actually falling.

So it appears there's a polarity issue related to the pull-up/downs, as well as a possible problem w/ accurately interrupting based on active-low signals (based on the erroneous interrupts reported on both edges above).

andyw-lala, 2015-09-25:
Maybe I'm missing something, but it seems like you're always configuring an
internal pull up/down in contention with your external pull up/down - I'm
not saying that's bad, just confusing. Is there a reason you're doing this ?

pomplesiegel, 2015-09-25:
Yes, they're intentionally in contention, as each one is a side of the SPST switch, so the state change occurs when the switch is closed.

for example, 
Left = 5V due to pull-ups and RC
Right = 0V at Photon input due to pull-down

Shouldn't `attachInterrupt()` be totally detached from whatever pull-up/down settings are chosen? If I can induce a signal at the uC's digital input which is clearly a falling edge (as measured on a scope), shouldn't the code trigger off of that edge, irrespective of pull-up/down settings?

andyw-lala, 2015-09-25:
But surely the switch is shorting one or other of the pull up/downs when it
is closed ?
So in the example above the 5V would be due to the switch, not the pull up.

I have a slight suspicion that the particle code might pay some attention
to the pull up/down you have configured, which is why I asked about testing
it without the on-chip pull up/down. Of course, this will require that your
external pull up/down goes to the opposite rail than the capacitor, and
that the switch be placed across the capacitor.

pomplesiegel, 2015-09-26:
Ah! I apologize for the false alarm. Basically, the discrepancy was caused by a deep setting in my scope causing an inversion on this one channel. Closing this now with a walk of shame :/ Thanks for your help!

closed 2015-09-26 by pomplesiegel

andyw-lala, 2015-09-26:
We've all been there.

I still think your setup of opposing pull up/downs is a little unusual.

pomplesiegel, 2015-09-26:
Haha Indeed.
The motivation for pull-ups + downs was to have the button be
* Active low
* Move from 5V (not 3.3) -> 0V upon a button press.
How else could this be accomplished?