particle-iot/device-os #645: Inconsistend relationship between PUBLIC/PRIVATE Particle.publish()/subscribe()

ScruffR, 2015-09-27:
While testing the fix for issue #371 I came across some aparent inconsistency between
Particle.publish() PUBLIC/PRIVATE and Particle.subscribe() PUBLIC/MY_DEVICES/DEVICE_ID over different firmware versions (at least for the Core 0.3.4 vs. 0.4.5 - Photon not yet tested).

It might be good to define what the actual relationship should be (set up a matrix), document this and have all versions act upon this standard.

My test code was this
```
void setup()
{
    Serial.begin(115200);
    //Particle.subscribe("publicScruff", handler);
    //Particle.subscribe("privateScruff", handler, MY_DEVICES);
    //Particle.subscribe("ScruffyMe", handler, "<myDeviceID>");

    Spark.subscribe("publicScruff", handler);
    Spark.subscribe("privateScruff", handler, MY_DEVICES);
    Spark.subscribe("ScruffyMe", handler, "<myDeviceID>");
}

const int cases = 6;
int i;
void loop()
{
    char nr[24];

    i++;
    sprintf(nr, "%5d < %5d >",i % cases, i);
    switch(i % cases)
    {
        case 0:
            // public event triggering public subscription
            Spark.publish("publicScruff", nr);
            break;
        case 1:
            // private event triggering MY_DEVICES subscription
            Spark.publish("privateScruff", nr, PRIVATE);
            break;
        case 2:
            // public event from my device should trigger device subscription
            Spark.publish("ScruffyMe", nr);
            break;
        case 3:
            // private event from my device should trigger device subscription
            Spark.publish("ScruffyMe", nr, PRIVATE);
            break;
        case 4: 
            // public event from my device should trigger MY_DEVICES subscription 
            // (even if publicly published)
            Spark.publish("privateScruff", nr);
            break;
        case 5: 
            // private event should not trigger public subscription
            Spark.publish("publicScruff", nr, PRIVATE);
            break;
        default:
            break;
    }
    
    delay(1000);
}

void handler(const char* event, const char* data)
{
    char str[256];
    sprintf(str, "%s: %s", event, data);
    Serial.println(str);
}
```

m-mcgowan, 2016-01-12:
After talking with the Cloud team, I have this information to bring to bear on the results:

- subscriptions to a specific device is presently not supported.
- subscriptions to MY_DEVICES receive only PRIVATE events. 

| publish\subscribe | ALL |  MY_DEVICES | Specific Device |
| --- | --- | --- | --- |
| PUBLIC | Y |  - | - |
| PRIVATE | - | Y | - |
------------------------------------------------

ScruffR, 2016-01-12:
I guess this is the ***planned*** behaviour, because at least when I did my tests above this was not entirely the case for on Core ***and*** Photon

m-mcgowan, 2016-01-12:
I'm surprised there is a difference between 0.3.4 and 0.4.x, since this is very much a cloud-side thing. And naturally there should be no differences between Core/Photon.  I will be extending our automated tests to cover these points so that we can be certain the actual behavior matches what is expected.

ScruffR, 2016-01-13:
Yup there is (at least was) as elaborated in the original issue #371

m-mcgowan, 2016-01-19:
I've verified with automated tests that the behavior on 0.4.x on the Photon is as expected (see the table a few comments up.)

m-mcgowan, 2016-01-19:
I wonder if there's a behavior difference between 0.3.4 and 0.4.x in the way events are processed - the test code above only calls delay(1000) which will pump the background events once. It would be better to avoid a delay() and use time elapsed since the last event so that the background tasks are serviced.

m-mcgowan, 2016-01-20:
The reason for the behavior difference on 0.3.4 is because 

```Spark.pubish("topic", "data", PRIVATE)``` 
didn't exist as an API until 0.4.x The actual api on 0.3.4 is

```Spark.publish("topic","data", ttl);```

Because an enum (PRIVATE) can be cast to an int (ttl) the compiler accepts the code. 

Changing the code to the 4-arg version produces consistent results on 0.3.4 and 0.4.7.

Thanks for the test code - it was a useful starting point. I tweaked your test code:
- starts with step 0
- executes a step when 's' is pressed, and outputs the step name
- runs the background loop otherwise
- private events published using the 4-arg API so it's compatible with 0.3.4 and 0.4.x 

```
void setup()
{
    Serial.begin(115200);
    //Particle.subscribe("publicScruff", handler);
    //Particle.subscribe("privateScruff", handler, MY_DEVICES);
    //Particle.subscribe("ScruffyMe", handler, "<myDeviceID>");

    Spark.subscribe("publicScruff", all_handler);
    Spark.subscribe("privateScruff", mine_handler, MY_DEVICES);
}

const int cases = 6;
int i;
void loop()
{
    char nr[50];

    char c = Serial.read();
    if (c!='s')
        return;
    sprintf(nr, "%5d < %5d >",i % cases, i);
    Serial.println(nr);
    switch(i % cases)
    {
        case 0:
            // public event triggering public subscription
            Spark.publish("publicScruff", nr);
            break;
        case 1:
            // private event triggering MY_DEVICES subscription
            Spark.publish("privateScruff", nr, 60, PRIVATE);
            break;
        case 2:
            // public event from my device should trigger device subscription
            Spark.publish("ScruffyMe", nr);
            break;
        case 3:
            // private event from my device should trigger device subscription
            Spark.publish("ScruffyMe", nr, 60, PRIVATE);
            break;
        case 4: 
            // public event from my device should trigger MY_DEVICES subscription 
            // (even if publicly published)
            Spark.publish("privateScruff", nr);
            break;
        case 5: 
            // private event should not trigger public subscription
            Spark.publish("publicScruff", nr, 60, PRIVATE);
            break;
        default:
            break;
    }
    i++;
    
}

void all_handler(const char* event, const char* data)
{
    char str[256];
    sprintf(str, "all: %s: %s", event, data);
    Serial.println(str);
}

void mine_handler(const char* event, const char* data)
{
    char str[256];
    sprintf(str, "mine: %s: %s", event, data);
    Serial.println(str);
}
```

The results on both 0.3.4 and 0.4.7 are:

```
    0 <     0 >
all: publicScruff:     0 <     0 >
    1 <     1 >
mine: privateScruff:     1 <     1 >
    2 <     2 >
    3 <     3 >
    4 <     4 >
    5 <     5 >
```

Which is as expected from the table above -  that only public events are picked up by an all events handler, and only private events picked up by a MY_DEVICES handler. 

Closing this out now that we've resolved the discrepancy between 0.3.4 and 0.4.7, documented the expected behavior and verified it.

closed 2016-01-20 by m-mcgowan