particle-iot/device-os #650: When calling analogWrite() at a rate near or above 500Hz PWM fails

ScruffR, 2015-09-30:
As in these threads discussed PWM puls-pause-ration gets corrupted by setting new values for PWM.
 https://community.particle.io/t/can-i-start-mirroring-the-led-color-to-my-own-led/15417/36
 https://community.particle.io/t/solved-system-mode-manual-messes-up-pwm-based-motor-speed-control/15735/7

and in slack https://s.slack.com/archives/elite/p1442869588001536 (+following)

When setting a new PWM value, don't re-set the pin, but latch the new value for activation ***AFTER*** the running PWM-cycle has finished (see slack discussion with HW hints).

cmsunu28, 2015-11-10:
Hey @ScruffR is there any way you can link the info without putting a slack link? Not everyone has access to slack.

ScruffR, 2015-11-10:
Sure, sorry!

jonlogan, 2015-11-10:
Just following up on Christine's request -- could you paste the relevant details into the issue here?   Thanks!

ScruffR, 2015-11-10:
Excerpt of slack discussion
[quote]
...
on a STM32 forum I found this
https://my.st.com/public/STe2ecommunities/mcu/Lists/STM32Discovery/Flat.aspx?RootFolder=https%3a%2f%2fmy%2est%2ecom%2fpublic%2fSTe2ecommunities%2fmcu%2fLists%2fSTM32Discovery%2fChanging%20of%20PWM%20duty%20cycle%20in%20an%20efficient%20manner&FolderCTID=0x01200200770978C69A1141439FE559EB459D75800084C20D8867EAD444A5987D47BE638E0F
With the suggestion: ​"You don't re-initialize and restart the timer to change the PWM.  You just write the updated width to the TIMx_CCRy register"_
...
You'll want to look at the ARPE bit in TIMx_CR1, with ARPE=0 the value for ARR is not shadowed. Also TIMx_EGR may be used to generate an update event, causing the preload values to be latched into the actual registers.
...
Each timer has an "auto reload" register which affects all that timer's channel counters (4).  The TIMx_ARR register is used to reload a timer's TIMx_CNT counter which can be configured for various modes (up, down, etc.).  To modify PWM, you just need to change the TIMx_ARR count which affects the clock to the timer channels' capture/compare registers.
...
So the auto-reload register (ARR) loads the counter (CNT) register which defines the clock to the channel capture/compare registers (CCR). 
...
[/quote]

peterqu, 2015-11-11:
Thank you #SccruffR for the excerpt.  I have already reproduced the error in a new unit test, and is now working on a solution.

monkbroc, 2015-12-05:
@peterqu fixed this with #752. I made sure my changes in #756 doesn't break this again. You can close this issue.

closed 2016-01-28 by m-mcgowan