particle-iot/device-os #653: OctoberY2k15 sleep bug (passing from end of month threshold to new month start)

jonlogan, 2015-10-01:
Observed: Customer reported that Photons that were sleeping during the boundary between September 30 2015 and October 1 2015 did not wake from sleep.  (This did not affect Cores).

Expected: Photons should be able to cross any date threshold and continue to sleep and wake from sleep as expected.

Workaround: a simple 'reset' of the device should resurrect a sleeping Photon, and will continue functioning until an upcoming month-end threshold.  (Fix for this will be coming asap)

jonlogan, 2015-10-01:
Customer's test to confirm the issue:  I just took a unit and set the rtc to 09-30 23:53 then with a loop of reporting and sleeping for 60 seconds. It stopped reporting at 09-30 23:59. So I think that confirms it. When the device awakens the internal led flashes white for a short duration none of that after 23:59

m-mcgowan, 2015-10-01:
fixed in commit https://github.com/spark/firmware/commit/4a61d52e2c2f743794d4a1b43bfa14963e58c711

closed 2015-10-01 by m-mcgowan