particle-iot/device-os #657: Particle.variable supports `String`s

m-mcgowan, 2015-10-02:
Using `Particle.variable()` is a confusing aspect of the Firmware API for coders not familiar with how C-strings work. 

https://community.particle.io/t/problem-with-the-new-release-v-0-4-6/16151

The `String` class was created to try to remove some of the confusion, but this isn't supported with `Particle.variable()`.  With support for String added, we can then write simple code like:

```
String variable;

void setup()
{
     Particle.variable("message", variable, STRING);
}

void loop()
{
     static int count = 0;
     message = String::format("on loop number %d", count++);
}
```

m-mcgowan, 2015-10-02:
Additionally, we can extend the API to take a reference rather than a pointer. Then all API calls look the same:

```
String s; 
int i;
double d;

Particle.variable("int", i, INT);  // note no & needed
Particle.variable("double", d, DOUBLE); // no & needed
Particle.variable("string", s, STRING); // no & needed
```

```

monkbroc, 2015-10-02:
> We could even avoid the last parameter, since it can be inferred from the type of the variable.

Great idea.

andyw-lala, 2015-10-02:
Is there a catchall method we could provide with a human readable #error to
respond to cases where the object passed, or the manner in which it was
passed does not fit one of the available methods ?

monkbroc, 2015-10-02:
One way would be to have a template that displays a a human readable compiler error for any template arguments, then specialize the template for all acceptable types with the correct code. It's a little verbose, but worth it from the user experience point of view.

wgbartley, 2015-10-05:
I know this is flagged for 0.4.7, but this worked with versions prior to 0.4.6.  Or if it didn't, it didn't cause any compile errors.  I just tried to update some older code tonight, and it no longer compiles (using CLI).  The error is:

```
../wiring/inc/spark_wiring_cloud.h:73:9: error: static assertion failed: 

In Particle.varible("name", myVar, STRING); myVar must be declared as char myVar[] not String myVar
```

However, the code it's griping about has not changed in months (since Maker Faire 2015?).  I have a variable defined globally as `String dhtError = ""` and in `setup()` I have `Particle.variable("dhtError", &dhtError, STRING);`.  I swear on my ancestors' graves that this compiled many many many times over.  In my particular use case, I don't need or use that variable anyways, so I commented it out.

Also, "varible" is typo'd in the error message.  I'd submit a PR for it, but I couldn't find "varible" in a GitHub search.

technobly, 2015-10-05:
Yeah it wouldn't have caused a compiler error before because it was not type safe (thus it allowed it anyway), and it would have worked until you changed the value of your String... at least I think I read that somewhere ;-)

wgbartley, 2015-10-05:
Ignorance is bliss

technobly, 2015-10-05:
Hey will this idea to make Strings compatible with Particle.variable()'s also make them work with Backup RAM (retained variables)? They currently don't because of the memory management.

m-mcgowan, 2015-10-05:
Sadly not brett. Strings won't work since they use dynamic memory allocated on the heap, which won't presently be persisted during sleep. We could possibly make a new heap in the backup RAM, but doing this is beyond my current knowledge - I don't know if the C runtime supports having multiple heaps.  

The plan for string support was going to be part of a generic callback mechanism for variables. For strings, the callback would provide `String.c_str()`. Other callback types would allow variables to be computed on demand, much like a function call.  Initially this would be an internal only change until we figure if this adds value or confusion as a user-visible API.

monkbroc, 2015-10-05:
> Other callback types would allow variables to be computed on demand

This will be popular! In many applications string variables are updated all the time yet only requested once in a blue moon.

technobly, 2015-10-05:
Oh man, I'm already thinking about how to abuse this new callback.  I think it violates the HTTP standards for GET being read-only.  But I so want to do it!  Doesn't seem right though... thoughts?

monkbroc, 2015-10-05:
Be strong! Resist the temptation to break the command-query separation!

m-mcgowan, 2015-10-05:
Variables are already subject to change, so we are not opening up any more power than what is already available, just making it more convenient for firmware to provide a variable's value via a callback when required.  Again, let me stress this would be an internal-only change, until we can work through all the use cases of exposing this as an API.

technobly, 2015-10-05:
Sounds good about system firmware only by default, and if we do provide a opt-in user api it should be very clear that it has a risk associated with it.  A simple browser GET request to the device will then possibly be able to modify data on the device.

commit 5e6c3a20 (2015-10-20): support for `String` variables. Fixes #657
closed 2015-10-20 by m-mcgowan