particle-iot/device-os #659: Multithreading: Second life for Particle.process()

ScruffR, 2015-10-02:
As per this discussion https://s.slack.com/archives/elite/p1443824246000119

Multithreading might lead people (like myself ;-) ) to the thought that we no longer need to exit `loop()` since all the cloud stuff is done in a seperate thread. 
But this is not the case for `Particle.function()` code and `Particle.subscribe()` handlers since they are considered application code and not system code. 
And even calling `Particle.process()` doesn't help this (as it would have in `SYSTEM_MODE(MANUAL)`) since it (aparently) got degraded to a mere stub, not doing anything.

So I'd see two options to save confusion
Either (as also proposed by Doc) let `Particle.process()` live on to do the function/subscription handling
or (my preference)
have the function/subsription handlers become callback functions that can be directly hooked up to the system thread - this might require some measures for thread/context safety on the coders side, but should be doable (e.g. as a C# guy [`InvokeRequired`](https://msdn.microsoft.com/en-us/library/ms171728(v=vs.110).aspx) strategies)

m-mcgowan, 2015-10-02:
Thanks for this. Great feedback. We can totally make Particle.process() function as an application event pump so that the app thread chooses when to allow functions/subscriptions to come through. :+1:

m-mcgowan, 2015-10-02:
The reason we don't want to push app code on to the system thread is because both should be separate - the app leaves the system to it's thing and vice versa. There are also deadlock concerns.

A third alternative is to run the application callbacks on a 3rd thread. We're keeping this in the wings for now, because then it means everyone has to be come experts at multithreaded code since their application code and their callbacks are running in separate threads, so shared data has to be guarded to prevent concurrent access. If we do this it will be as an opt-in arrangement for folks that know what they are getting into (or have read our explanation of what they are getting into!)

technobly, 2015-10-03:
:+1: for a way to pump the function/subscription handlers (easy and intuitive), instead of needing to know loop() should end to allow that to happen.

ScruffR, 2015-10-03:
In connection with this I remembered this thread
https://community.particle.io/t/flush-messages-before-going-to-sleep/8902/8

where going to `System.sleep()` interfered with pending `Particle.publish()` and the workaround to call `Particle.process()` before going to sleep could get affected by the new `Particle.process()`.

Just to take into consideration too.

m-mcgowan, 2015-10-03:
Good point there ScruffR.  As well as pumping application events, Particle.process() can also wait for the system event queue to flush. I'll write a test for specifically this case - publish then sleep - to be sure the system functions as expected since it's definitely an edge case.

ScruffR, 2015-10-03:
It might not always be required to wait for the queue to be flushed but sometimes is, so how about this `Particle.process(bool flushQueues = true);`?
In this case working code won't break and the above workaround still works, but for speed-freaks there might be a few cycles to spare.

ScruffR, 2015-10-06:
Investigating workarounds for #670 and [this forum thread](https://community.particle.io/t/system-thread-during-listen-mode/16133), I tested `RGB.onChange()` in connection with `SYSTEM_THREAD(ENABLE)` and realized the `onChange()` handler is still working, even if appllication code is blocked (and during Listening Mode).

From a blue-eyed observers view I would say that the onboard RGB-LED is driven from a system routine, which would naturally be expected to be running in the system task and it also apears (and the name suggests) that the `onChange()` handler gets called from that system routine, since it's absolutely unimpressed by blocking user code.

But this would - for my understanding - not fit this quote from the slack discussion linked above
```text
However the `Particle.function` code and `Particle.subscribe` handlers ​*are*​ code you wrote — they run on the application task.
```
Since the `onChange()` handler is *also* code written by me, but still apears to be called and running as if it were called by the system.

So either `RGB.onChange()` might bear the same risks as outlined above for my proposal about having `Particle.function()` and `Particle.subscribe()` handlers and this would have to be clearly stated ***or*** my proposal might be back in the race ;-)

What do you think??

m-mcgowan, 2015-10-20:
That's a good point about onChange being application code. If we were being completely consistent then it would too be executed on the application thread. 

The rationale for keeping RGB.onChange() handler's called from whatever context is changing the LED is that you typically want these functions to execute immediately as the LED changes, and very quickly.  If we bounce the call to another thread, then there could easily be a few milliseconds delay.  

If a user really wants the LED handler to run on the application thread, we could add a way to build a function that runs on the application thread:

```
void MyRGBHandler(uint8_t r, uint8_t g, uint8_t b)
{
   Serial.printlnf("r: %d, g: %d, b:%d", r, g, b);
}

void setup()
{
   RGB.onChange(ApplicationHandler(MuRGBHandler));
}
```

ScruffR, 2015-10-20:
I guess you actually meant "system thread" here
[quote]
That's a good point about onChange being application code. If we were being completely consistent then it would too be executed on the application thread.
[/quote]

And yes, I agree!

And "YES!" again, that application handler suggestion sounds great.

wgbartley, 2015-10-20:
What are the common use cases for RGB.onChange()?  So far, it seems to be to "mirror" the onboard LED.  Perhaps you could mitigate user code running amuck by exposing the 3 RGB values as system or reserved variables that can only be get (not set) by user code?

Disregarding the variable idea, I agree with ScruffR about the application context.  You can really do some damage trying to "mirror" the LED with 300 NeoPixels inside the event handler!

ScruffR, 2015-10-20:
@wgbartley, I've been so naughty once to abuse the `onChange()` handler to timeout an activated Listening Mode without the need for `IntervalTimer` ;-)

m-mcgowan, 2016-01-29:
Implemented in 0.4.9 - `Particle.process()` acts as an application event pump in threaded and non-threaded modes.

closed 2016-01-29 by m-mcgowan