particle-iot/device-os #660: Multiple connections to TCPServer can cause the chip to crash

BrianGenisio, 2015-10-03:
We found a bug in TCPServer which causes the chip to restart.  We are using the basic "echo" [example from the docs](https://docs.particle.io/reference/firmware/photon/#tcpserver)

To reproduce:
1. telnet to the IP address
2. telnet to the IP address a second time.  Hold down the carriage return

This was found in a robot competition where someone wrote a script to disrupt the competition.  We've tried, but we can't figure out a way to modify that example to avoid the crash.

m-mcgowan, 2015-10-13:
I was unable to reproduce this. Could you please provide more details:

- what type of device (Core/Photon/Electron)
- what version of system firmware were you using
- is this always reproducible or just intermittently?

BrianGenisio, 2015-10-14:
Sorry, I missed some critical details that should have been in the report.  It is a Photon running version 0.4.6 of the firmware.  This behavior was observed in all the chips we used (about 10).  It is consistently reproducible.

It was found because we were running a robotics competition and VoodooSpark was loaded on the chip.  One of the contestants ran a Python script that just looped through all of the known IPs, connected to the port, and sent "Hello World" to the device.  After the second or third time through the loops, the chips would restart.  It was predictable enough that they did it every time.  Some chips even went into a permanent "safe mode" state where the only way I could get them back was by using dfu-util to load the firmware in two parts.

So we spent some time trying to reproduce it, and we were able to reproduce it predictably using the simple echo server example in the docs (eliminating VoodooSpark from the equation).

m-mcgowan, 2015-10-14:
Thanks for the additional detail. Could you please try against firmware 0.4.5. You'll first need to downgrade using dfu-util to flash the 0.4.5 binaries manually. Then compile your app in the WebIDE, selecting 0.4.5 as the target version for your device. 

I believe this issue has already been fixed in the develop branch, if you are able to use the local compile.

BrianGenisio, 2015-10-14:
Sure, I'll try that.  To be clear, you want me to try an older firmware, not a newer firmware?  Are you suggesting that the bug was introduced in 0.4.6, so it shouldn't exist in 0.4.5? 

Thanks,
B

m-mcgowan, 2015-10-14:
Yes, exactly that. I think it was a regression in 0.4.6 which we've fixed in preparation for 0.4.7

BrianGenisio, 2015-10-15:
I downgraded my firmware using [0.4.5](https://github.com/spark/firmware/releases/tag/v0.4.5) and built with firmware 0.4.5 and I'm still seeing the problem.  I took a [video of the problem](https://youtu.be/GcxqyMqshgU).

The steps were slightly different than before, but it still happened.

1. Load the demo to the chip
2. Connect via serial to start the server
3. Telnet into the server once
4. Telnet into the server twice
5. On second connection, start sending garbage
6. On the first connection send a couple of characters

At about 50 seconds, you will see the chip restart itself and my connection will be dropped.

m-mcgowan, 2016-01-29:
Thank you for the clear steps to reproduce the issue.

The echo server demo isn't a general server - it's intended as a simple guide to working with TCP server and consequently handles one connection at a time. Once the 2nd telnet connection is established, it is not in fact processed by the application as long as the first connection is still connected (`client.connected()` will continue to return true, so `server.available()` is not called to retrieve the 2nd connection.) With the socket not being serviced, the system runs out of network resources, causing the cloud connection to fail and be restarted. 

The chip isn't resetting entirely - your program code is still executing according to  the `SYSTEM_MODE()` setting. The cloud connection is taken down because the device didn't receive a response to the heartbeat that is sent to the server (presumably because there were no free packet buffers.) A new connection is attempted. When this doesn't succeed 2 times in succession, the WiFi network is restarted. This clears all the congestion from the packet buffers, allowing the device to reconnect.

BrianGenisio, 2016-01-29:
Thanks for the details.  Are you saying there is something I can do in the code to fix this problem?  I agree that the echo server isn't intended to be a general server.  I was just using it to illustrate the problem.  The bigger problem is that this problem was used to deliver a DoS attack on our chip.

The code that is running is called [VoodooSpark](https://github.com/voodootikigod/voodoospark), and it is used with [JohnnyFive](http://johnny-five.io/) as a subordinate host.  The loop code is [here](https://github.com/voodootikigod/voodoospark/blob/master/firmware/voodoospark.cpp#L1080).

The problem, in practice, is that we get our first connection up and running.  Everything is fine.  But if a nefarious agent connects up to the same IP:port and starts sending gibberish, the chip disconnects, and the original connection is dropped.  It doesn't take much to make this happen, so it becomes very easy for someone with ill will to attack the chip with the connection, thus disabling the robot.

Given the above loop code, is there something we can do to make this better?  We had tried accepting the second connection and closing it immediately, but IIRC, we never actually received the connection.  Something further down the stack prevented it, so we weren't able to come up with anything in VoodooSpark to avoid the DoS.

Bacichetti, 2016-12-16:
My application can receive multiple simultaneous client connections and it makes the device to crash. 

Our device/controller automates Blinds/shades/etc it's meant to be used with our own app OR a bridge to home automation systems (Control 4 for example). The way driver are implemented by a system like Control 4 is that each blind (or rooms with blinds) is a CLIENT. No problem, unless whoever install our controller makes a script where all blinds/shades try to connect and send commands at the exact same time.

On my tests I see that my Photon works well up to ~4 simultaneous client connections. More than that the device crashes (some times only losing wifi connection, some few times having "hard faults"). After Cloud reconnects, some of the clients messages are actually still "in memory / queue" and the commands are sent others are lost.

If client connection requests are spaced by e.g. 100 ms the number of connections it can manage increases to at least 2x that 4 initial connections, maybe more (even though my device needs about 500 ms to fully process each connection/message).

There are ways so that the script could just NOT try to connect and send commands at the same time using Control 4, that fix the problem. However, using other system that may not be possible.

Some questions I have:
How many simultaneous new requests for connections Photon should be able to manage?
How many simultaneous connections it can hold?
Does all the connections share the same buffer (128 bytes if I remember) or they get a buffer each?
Wouldn't it be better to deny new connections when the socket queue is full than just crash?

I'm running firmware version 6.0 for reference.

technobly, 2016-12-16:
Hi @Bacichetti thanks for the report.  It sounds like we could answer your questions and possibly uncover some lurking bugs in the process by adding a TCPClient / TCPServer test that looks to ensure stability by maxing out the allowable connections.  I'll add this to our internal issue tracker so we can come up with an answer to these questions.

Bacichetti, 2017-01-05:
Hi @technobly, any news about this issue?

technobly, 2017-01-05:
Hi @Bacichetti no not yet.  Much of the firmware team has been on holiday break and it will take some time before this gets prioritized.  We'll be sure to update this issue when there is progress :)

m-mcgowan, 2017-07-07:
Hey @Bacichetti - 0.7.0-rc.1 has been released which includes a number of TCP fixes. Please take it for a spin and let us know if the problem still persists or not! Thanks :-)

m-mcgowan, 2017-10-22:
Closing due to inactivity and that we believe the issue to have been fixed.

closed 2017-10-22 by m-mcgowan

Bacichetti, 2017-10-23:
Hi, @m-mcgowan, I missed your message to me, but I did test the new changes and I can tell the difference! I'm not using 0.7.0-rcX extensively, but the tests showed the TCP port is not crashing anymore for multiple simultaneous connections anymore.
Thanks, that's a huge improvement to our system stability!
Waiting for 0.7.0 release!