particle-iot/device-os #661: retained variables are not persisting, even without reset or deep sleep.

technobly, 2015-10-03:
Please see:
https://community.particle.io/t/backup-ram-not-working/16179

Just to confirm, is it correct that retained variables should come up as the last written value after a reset or deep sleep event, and not be re-initialized if declared and initialized globally?  So in the first two posts of the above thread, we have:

`retained int saveValue = 10;`

If the last written state is 500 lets say, after reset it would still be 500, yeah?

m-mcgowan, 2015-10-03:
Strange. I woke up thinking about this and then remembered that the docs don't include the command to enable retained variables. 

```
STARTUP(System.enableFeature(FEATURE_RETAINED_MEMORY));
```

m-mcgowan, 2015-10-03:
Docs updated.

technobly, 2015-10-03:
Is there a reason it's not always enabled?  Since it has a keyword it's pretty unique.

technobly, 2015-10-03:
I just tested with the enable command and not retained seems to work just like normal variables when powered, but if I reset when I'm in the 5 second delay it comes up setting the value to 10 again, instead of 15.  If I don't initialize the value, it still comes up as 0 instead of 15.

```cpp
STARTUP(System.enableFeature(FEATURE_RETAINED_MEMORY));

retained int saveValue = 10;

void setup() {
    Serial.begin(9600);
}

void loop() {
        Serial.println(saveValue);
        saveValue = 15;
        Serial.println(saveValue);
        Serial.println(saveValue);
        delay(5000);
}
```

technobly, 2015-10-03:
Ok figured this out... thought it was ok to hard reset it (pushing the reset button) but it only retains memory with a soft reset, E.g., System.reset();

I'll update the docs with that clarification ;-)

technobly, 2015-10-04:
Ahh, so much confuse... there was an issue with how SRAM was initialized which led to much confusion.  Fixed with this PR: https://github.com/spark/firmware/pull/664

closed 2015-10-20 by m-mcgowan