particle-iot/device-os #663: Photon frequently SOS's immediately following cloud re-connect

pomplesiegel, 2015-10-04:
Using system FW 0.4.6 and the code below, the Photon often crashes (red flashing SOS) immediately upon a cloud re-connect, following a WiFi disruption. 

I noticed this because our application which ran stable on 0.4.5 is often crashing under the same circumstance (re-connecting to the cloud, possibly while publishing?). Below is basically a representation of a normal application, in order to successfully induce the same failure. 

#### Steps to reproduce :
1. Load code below on a Photon (running system fw 0.4.6) w/ WiFi creds already installed. The photon will connect to a network (breathing cyan) and begin publishing
2. Remove the WiFi network (pull the plug on router or disable this network)
3. Wait a few seconds
4. Re-enable the network
5. Repeat steps 2-4 until the Photon SOS's. From my repeated experience this only takes 1-3 times. 

#### Things to note
1. This is SYSTEM_MODE(MANUAL) using the new SYSTEM_THREAD(ENABLED)
2. There is a delay in the program (mimicking a normal program's behavior)
3. We're constantly checking Particle.connected(), then publishing once that returns true. 
Given the changes in multi-threading, might this be a concurrency issue regarding this system fw -> user app state?

````cpp
#include "application.h"

SYSTEM_MODE(MANUAL);
SYSTEM_THREAD(ENABLED);

void setup()
{
  Serial.begin(9600);
  delay(2000);
  Serial.println("Beginning program!");

  WiFi.on();
  WiFi.connect();
  Particle.connect();
}

int timeOfPublish = 0;

void loop()
{
  //If we're connected, once per second
  if( Particle.connected() && ( timeOfPublish != Time.now() ) ) 
  {
    //Publish
    Serial.println("publishing...");
    Particle.publish("test","testData");
    timeOfPublish = Time.now();
  }

  //Manage WiFi/Cloud overhead
  Particle.process();
  
  //Delay of a normal program 
  delay(5);
}
````

HardWater, 2015-10-05:
I can confirm that I experienced the same SOS problem when trying to use 4.6 it occurred on the second http request  in a code loop (awake cycle) communicating with my server . The first request succeeds. This seems to be a bug that was fixed in 4.4 but has come back. I switched to 4.5 and was not able to replicate the SOS problem.

See https://community.particle.io/t/httpclient-and-the-photon/12661/22

nlambuca, 2015-10-05:
exactly. Will try the same... but how is it possible they wen back in time? Issues committing?

m-mcgowan, 2015-10-09:
This is fixed in develop. Much appreciated if someone is able to test and confirm.

indraastra, 2015-10-09:
It's my first time trying to build straight from the git repository and I'm running into this:

```vishal@foo:/particle-firmware/modules$ make PLATFORM=photon

/Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Users/vishal/Documents/Workspace/particle-firmware/modules/photon/system-part1/   PLATFORM=photon
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C ../../../communication 
make[2]: Nothing to be done for `all'.
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C ../../../hal-dynalib 
make[2]: Nothing to be done for `all'.
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C ../../../services 
make[2]: Nothing to be done for `all'.
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C ../../../platform 
make[2]: Nothing to be done for `all'.
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C ../../../hal 
make[2]: Nothing to be done for `all'.
make[1]: *** No rule to make target `../../../build/target/communication/platform-6-m-prod-6/libcommunication.a', needed by `../../../build/target/system-part1/platform-6-m/React.elf'.  Stop.
make: *** [/Users/vishal/Documents/Workspace/particle-firmware/modules/photon/system-part1/makefile] Error 2```

indraastra, 2015-10-09:
Whoops, I accidentally deleted my previous post. I'm not sure if this is anything to do with your recent work since it's my first time building from scratch, but I'm running into this error after a whole bunch of files successfully compile:

```
$ make clean all program-dfu PLATFORM=photon
....
Building file: ../../../modules/photon/system-part1/src/wifi_dynalib.c
Invoking: ARM GCC C Compiler
mkdir -p ../../../build/target/system-part1/platform-6-m/../../../modules/photon/system-part1/src/
arm-none-eabi-gcc -DSTM32_DEVICE -DSTM32F2XX -DPLATFORM_THREADING=1 -DPLATFORM_ID=6 -DPLATFORM_NAME=photon -DUSBD_VID_SPARK=0x2B04 -DUSBD_PID_DFU=0xD006 -DUSBD_PID_CDC=0xC006 -DPRODUCT_ID=6 -DPRODUCT_FIRMWARE_VERSION=65535 -DUSE_STDPERIPH_DRIVER -DDFU_BUILD_ENABLE -fno-builtin -DSYSTEM_VERSION_STRING=0.4.6 -DRELEASE_BUILD -Werror -I../../../modules/photon/system-part1/inc -I../../../modules/shared/stm32f2xx/inc -I../../../communication/lib/tropicssl/include -I../../../communication/src -I../../../services/inc -I../../../platform/shared/inc -I../../../platform/MCU/STM32F2xx/CMSIS/Include -I../../../platform/MCU/STM32F2xx/CMSIS/Device/ST/Include -I../../../platform/MCU/STM32F2xx/SPARK_Firmware_Driver/inc -I../../../platform/MCU/shared/STM32/inc -I../../../platform/MCU/STM32F2xx/STM32_StdPeriph_Driver/inc -I../../../platform/MCU/STM32F2xx/STM32_USB_Device_Driver/inc -I../../../platform/MCU/STM32F2xx/STM32_USB_Host_Driver/inc -I../../../platform/MCU/STM32F2xx/STM32_USB_OTG_Driver/inc -I../../../hal/inc -I../../../hal/shared -I../../../hal/src/photon -I../../../hal/src/stm32f2xx -I../../../hal/src/stm32 -I../../../hal/src/photon/api -I../../../dynalib/inc -I../../../rt-dynalib/inc -I. -MD -MP -MF ../../../build/target/system-part1/platform-6-m/../../../modules/photon/system-part1/src/wifi_dynalib.o.d -ffunction-sections -fdata-sections -Wall -Wno-switch -Wno-error=deprecated-declarations -fmessage-length=0 -fno-strict-aliasing -DSPARK=1 -DPARTICLE=1 -DSTART_DFU_FLASHER_SERIAL_SPEED=14400 -DSTART_YMODEM_FLASHER_SERIAL_SPEED=28800 -fno-builtin-malloc -fno-builtin-free -fno-builtin-realloc -DUSER_FIRMWARE_IMAGE_SIZE=0x20000 -DUSER_FIRMWARE_IMAGE_LOCATION=0x80A0000 -DMODULAR_FIRMWARE=1 -DMODULE_VERSION=7 -DMODULE_FUNCTION=4 -DMODULE_INDEX=1 -DMODULE_DEPENDENCY=0,0,0 -g3 -gdwarf-2 -Os -mcpu=cortex-m3 -mthumb -Wno-pointer-sign -std=gnu99  -c -o ../../../build/target/system-part1/platform-6-m/../../../modules/photon/system-part1/src/wifi_dynalib.o ../../../modules/photon/system-part1/src/wifi_dynalib.c

make[1]: *** No rule to make target `../../../build/target/communication/platform-6-m-prod-6/libcommunication.a', needed by `../../../build/target/system-part1/platform-6-m/React.elf'.  Stop.
make: *** [/Users/vishal/Documents/Workspace/particle-firmware/modules/photon/system-part1/makefile] Error 2
```

m-mcgowan, 2015-10-09:
Check you don't have APP or APPDIR defined in your environment since that seems to be affecting the build. I've no idea why it's labelling the output as React.elf.

indraastra, 2015-10-09:
`env | grep React` told me I had some suspiciously named variables set to "React", like TARGETNAME and TARGET_NAME (probably by Xcode or a previous build?). Unsetting TARGET_NAME and rebuilding did the trick, but I have no idea why it doesn't get overridden by the makefiles. Then again, I don't really understand makefiles :smile: 

I will report back with a more relevant update shortly.

indraastra, 2015-10-09:
Ok, so I'm not running exactly the same code as @pomplesiegel, but rather, a version of the code user mhazley posted here:

https://community.particle.io/t/failures-on-handling-wifi-reconnects-v0-4-5-v0-4-6/16384

I can confirm that killing the AP no longer causes an immediate SOS from the `Particle.connect()` that follows once the Photon realizes it's no longer connected. 

That said, I followed @pomplesiegel's steps 3 times without power cycling and the Photon reconnected to the cloud successfully the first two times and the SOS'd the third time.

indraastra, 2015-10-09:
I repeated the experiment with the exact code posted above and can confirm that it's still happening on the third reconnection. Not sure what it is about the third time, but I noticed the LED flash red momentarily on the second reconnect.

On that note, why does the Photon even attempt to reconnect to the cloud after a disconnect in this code? Is it the call to Particle.process()?

m-mcgowan, 2015-10-10:
Flashing RED (or orange as it is in the latest code) momentarily is fine - that can be due to a failed connection attempt. It's not a permanent error and is expected in some situations. (I changed it from red to orange to avoid confusion with SOS.)

m-mcgowan, 2015-10-10:
`Particle.connect()` sets a flag that says "connect to the cloud" - the system will continue attempting to connect to the cloud whenever loop exits or `Particle.process()` is called (in single-threaded mode.) In multithreading, the system will continually attempt to connect on the background thread.  This continues until the connection succeeds or `Particle.disconnect()` is called.

mhazley, 2015-10-10:
Thanks - should hopefully get a test run on this today.

mhazley, 2015-10-10:
@m-mcgowan I tested that earlier and got no SOS - looks like you got that. 

Have you got a PR or commit for that fix anywhere? I'm not sure I want to run Develop in the field this week so I might run a local 0.4.6 with that fix until 0.4.7 comes out. Or will that fix be going onto the 0.4.6 release branch for an 0.4.6.2?

indraastra, 2015-10-12:
I can also happily report that the SOSes I was seeing on Friday are no longer happening in the latest develop branch. Thanks!

indraastra, 2015-10-12:
@m-mcgowan Ah, so my takeaway from your explanation of `Particle.connect()` is that once called, you never need to call it again for the cloud connection to be re-established after a disconnect or any kind of connection upset?

m-mcgowan, 2015-10-12:
Yes, that's correct, although in manual mode you also are responsible for pumping messages with `Particle.process()`

indraastra, 2015-10-12:
Except if threading is enabled, in which case `Particle.process()` is called on the system thread?

m-mcgowan, 2015-10-12:
Correct. With threading enabled, the only distinction between the modes is if the system starts with the cloud connected or not. The system implicitly does a Particle.connect() on startup when the mode is automatic.

indraastra, 2015-10-12:
Great, thanks for the clarifications! I'm trying to update my wifi bringup and monitoring code to be more in line with all these details, and eagerly (but patiently) await the fix you have in store for the `hasCredentials()` reentrancy issue.

pomplesiegel, 2015-10-13:
@m-mcgowan, interesting! Given that insight, when threading is enabled, is there any difference between `SEMI_AUTOMATIC` and `MANUAL`? In #677 we're seeing different behavior between the two, as functions are not callable in `MANUAL` without adding `delay(1)` to loop()

m-mcgowan, 2015-10-13:
In principle, no difference. The issue you're seeing may be the result of thread priorities preventing execution of the system thread without the call to delay() which then allows the system to perform a thread yield. I will be investigating that in the coming days.

pomplesiegel, 2015-10-13:
Great! I figured it could be something like that - the OS thinking that this thread is not significant enough to run at our anticipated frequency. 

Thanks for looking into this!

closed 2015-10-20 by m-mcgowan