particle-iot/device-os #666: [Photon] `Wire` changes in 0.4.6 causing more frequent i2c errors on MCP23017

pomplesiegel, 2015-10-05:
Following the `Wire` (i2c) changes from PR #626 integrated into FW 0.4.6, we're seeing far more frequent bus hangups (SDA being held low by the MCP23017) than in 0.4.5. 

Just as before, the issue occurs in the MCP23017 driver method `readRegister()` wherein the Photon attempts to read from the MCP23017, calling

````cpp 
Wire.endTransmission();
Wire.requestFrom(i2caddr, (uint8_t)1
````

At this point the bus gets hung up, causing `Wire.requestFrom()` to return `0`, a condition which the system holds indefinitely.

Just as before, `Wire.reset()` (or a homegrown version of this code) frees the bus of this condition, but it is unnerving for this to happen even more often than before the i2c changes.

Is anyone else seeing this?

technobly, 2015-10-05:
@pomplesiegel do you think you can provide a bare minimum example with your readRegister() function so that we can reproduce easily?  I have a MCP23017 that I can try this on, but it would be a big help to start with the exact code you are using.

As a shot in the dark, bus hangups sound like maybe the MCP23017 is not ready to clock out data yet.  It might be worth adding a `delay(5);` after the endTransmission(); calls.  

If you have any other specific notes about expected behavior vs actual behavior with a logic analyzer screenshot, that would be most helpful.

pomplesiegel, 2015-10-05:
@technobly, thanks for the response. I'll begin working on a an example to try to lock up the bus after repeated reads.

The `readRegister()` method is from [this well known Adafruit driver for the MCP23017] (https://github.com/adafruit/Adafruit-MCP23017-Arduino-Library). 

Your hypothesis may be correct regarding the MCP23017 not being ready to clock out data! This is often the case with some i2c sensors which can sometimes take a considerable amount of time before being ready to deliver data. However,  a delay like `delay(5)` is quite a lot for real-time applications. Is there any way for us to know if the MCP23017 is actually ready before issuing the readRegister() command, which from our experience likely causes the bus latch-up?

My concerns are that
1. I believe the Wire.requestFrom() failures are happening more often using this new i2c implementation, which is specifically in place to fix issues w/ the MCP23017 chipset. 
2. This MCP23017 issue has quite a [long history] (https://community.particle.io/t/i2c-lcd-display-getting-corrupted-solved/9767/95) with both the Spark Core and Photon, showing as LCD corruption/bus lock-up behavior. 
3. Even with 50uS delays in our code I have seen this bus lock-up still occur after the program running for over a day. Once again, this is solved by a call to Wire.reset(), but a bit scary!

m-mcgowan, 2015-10-16:
The forum post indicates this is an issue with `SYSTEM_THREAD(ENABLED)`. Are we ok closing this one out?

closed 2015-10-16 by m-mcgowan

pomplesiegel, 2015-10-16:
Sure. Planning to gather more data regarding `SYSTEM_THREAD(ENABLED)`'s effect on i2c. We're repeatedly only seeing i2c data corruption (LEDs incorrectly turning on/off on a PE) when running with `SYSTEM_THREAD(ENABLED)`. When this line is commented, the problem disappears. 
@m-mcgowan, any thoughts on this?

m-mcgowan, 2015-10-16:
I'm guessing we need to disable task switching at key places in the code to preserve critical timing loops. I'm kind of surprised we haven't seen failures before since WICED is always task switching, but with multithreading we are leaning on task switching much more heavily and fully exposing the problem.

pomplesiegel, 2015-10-16:
Interesting! Might it be enough to disable task switching within the body of [`HAL_I2C_End_Transmission`] (https://github.com/spark/firmware/blob/aefb3342ed50314e502fc792f673af7a74f536f9/hal/src/stm32f2xx/i2c_hal.c#L394)? If that's an easy thing to try, with your help I could certainly make the change/compile locally and give it a shot.

m-mcgowan, 2015-10-16:
Sure, the call is 

```
#include "concurrent_hal.h"

os_thread_scheduling(false, NULL);  // turn off thread scheduling
os_thread_scheduling(true, NULL);  // turn on thread scheduling
```

The HAL_I2C_End_Transmission function has lots of exits, so to make this simple, I would rename that function to end_transmission(), then add this code:

```
uint8_t HAL_I2C_End_Transmission(HAL_I2C_Interface i2c, uint8_t stop, void* reserved)
{
    os_thread_scheduling(false, NULL);
    uint8_t result = end_transmission(i2c, stop, reserved);
    os_thread_scheduling(true, NULL);
    return result;
}

```

Cheers!