particle-iot/device-os #671: analogWrite to DAC pins on Photon no longer works after upgrading to 0.4.6

reinerge, 2015-10-06:
After upgrading to 0.4.6 analogWrite on photon no longer outputs the given DA value but instead outputs 3.3V
Repro steps: 
1.   run code below on a photon device with 0.4.5 firmware and watch DAC output pin using a scope. It should show a sawtooth waveform signal.
2.  cloud compile code for 0.4.6 and flash device. Device update to 0.4.6 starts. 
3.  Wait for update to 0.4.6 firmware to finish. Scope shows 3.3V constant signal. If there's still a sawtooth visible, re-power device.

    void setup() {

    }

    void loop() {
        for(int i = 0; i < 4096; i++) {
            pinMode(DAC, OUTPUT);
            analogWrite(DAC, i);
        }
        delay(100);
    }

reinerge, 2015-10-06:
**Update:**
I tried to find out what could be the reason for that issue and in order to do so I started to add some Serial.print's inside the top level function analogWrite in the 0.4.6 firmware code just to see what is called and what is not (I switched to local build for that, syncing to the 0.4.6 repository on my machine).

**So here's the surprise:**
adding Serial.println() between the safety check and the call to HAL_Validate_Pin_Function made the analogWrite function work again.
adding a simple delayMicroseconds(0) instead does the same. Adding the delay function at the beginning of the analogWrite function does however not work.
This smells like voodoo...

In addition, with this change it is no longer necessary to call pinMode(DAC, OUTPUT) everytime you call analogWrite.

Here's my modified analogWrite function:

    void analogWrite(pin_t pin, uint16_t value)
    {
        // Safety check
        if (!pinAvailable(pin))
        {
            return;
        }
        delayMicroseconds(0);
        if (HAL_Validate_Pin_Function(pin, PF_DAC) == PF_DAC)
        {
            HAL_DAC_Write(pin, value);
        }
        else if (HAL_Validate_Pin_Function(pin, PF_TIMER) == PF_TIMER)
        {
            PinMode mode = HAL_Get_Pin_Mode(pin);

            if (mode != OUTPUT && mode != AF_OUTPUT_PUSHPULL)
            {
                return;
            }

            HAL_PWM_Write(pin, value);
        }
    }

I would guess that there has been an issue with analogWrite also in the previous releases that was hidden by the call to pinMode before. Now some timing or whatever has changed and that "trick" does not work anymore -- means the issue is somewhere in the HAL code.
The modification works for both my production device and the test device.

ScruffR, 2015-10-26:
Issue #662 gets superceeded and worsened by this one.

pingdynasty, 2015-10-27:
I think these lines should be appended to `HAL_DAC_Init()` in dac_hal.c:
```
  DAC_Cmd(DAC_Channel_1, ENABLE);
  DAC_Cmd(DAC_Channel_2, ENABLE);
  DAC_SoftwareTriggerCmd(DAC_Channel_1, ENABLE);
  DAC_SoftwareTriggerCmd(DAC_Channel_2, ENABLE);
```
and `DAC_Cmd(...)` should be removed from `HAL_DAC_Write()`, which will fix #662.

like so:
https://github.com/marsus/firmware/commit/bd6e4678d9e60dad569bbc02875ee30bef1e5914

unreason, 2015-11-03:
The bug goes away for me if I set the remaining unused field in DAC_InitStructure to zero - (DAC_InitTypeDef::DAC_LFSRUnmask_TriangleAmplitude). This could explain why the problem gets fixed with the delayMicroseconds(0) call - which might set up the stack in a way that the uninitialized field has the correct value.

Can someone else please try this and confirm if it fixes the problem for you too?

reinerge, 2015-11-03:
Works for me on firmware 0.4.6

jcard0na, 2015-11-09:
I just built from the developer branch and this works for me, but only for DAC1 (pin DAC) and not DAC2 (pin A3).  Also, it looks like the unit test that was added in commit 4063e99edc94b5535422bf7424b847f55d1ed3ee only tests for DAC1:
```diff
+test(ADC_AnalogReadOnPinWithDACOutputResultsInCorrectValue_2) {
+    pin_t pin = A5;
+    // when
+    // Github issues #671
+    for(int i = 0; i < 4096; i++) {
+        pinMode(DAC, OUTPUT);
+        analogWrite(DAC, i);
+    }
+    delay(100);
+    analogWrite(DAC, 2048);
+    int32_t ADCValue = analogRead(pin);
+    // then
+    assertTrue((ADCValue>2000)&&(ADCValue<2100));
+}
```
May I suggest you also test for DAC2 before marking this as fixed?

Cheers,

reinerge, 2015-11-10:
I just picked the changes in dac_hal.c into a 0.4.7 release firmware, added string.h to the list of includes (for memset) and both analog outputs work. There seems to be no need to call pinMode anymore.
```
#include <application.h>
SYSTEM_THREAD(ENABLED);

void setup() 
{
}

void loop() {
    for (int i = 0; i < 4096; i++) {
        analogWrite(DAC, 4095 - i);
        analogWrite(DAC2, i);
    }
}
```

peterqu, 2015-11-10:
Hi @jcard0na, I just did some more testing on DAC2.  DAC2 analog output works for me.  Can you redo your test and let me know if you are still running into problem with DAC2?

Thank you for the reminder, I will add DAC2 to the unit test :-).

jcard0na, 2015-11-10:
Hi Peter,

I had forgotten to set the pinMode on DAC2 when I last tested it *blush*.
I just re-tested and works well.

Thanks, and sorry for the noise.

Javier

hboyce4, 2015-12-04:
Thank you @reinerge  for the workaround! Saved our asses!

jmaxxz, 2015-12-17:
Encountering same issue. This is a pretty big problem, can we get this fixed sooner than 0.4.9?

peterqu, 2015-12-17:
Hi @jmaxxz 
Release 0.4.8 and the develop branch include a fix for this issue.  Can you double check if to see if you are running the latest develop branch or release 0.4.8 system firmware?

dmiddlecamp, 2015-12-17:
re: @peterqu -- we haven't released 048 yet right?

peterqu, 2015-12-17:
@dmiddlecamp , you are right!  We haven't released 0.4.8 yet, we only have 0.4.8 rc1.
@jmaxxz , we already have a fix for this issue and it will be included in the next release.  If you want to get it sooner, you can compile from develop branch.

m-mcgowan, 2016-01-18:
Fix verified.

closed 2016-01-18 by m-mcgowan