particle-iot/device-os #672: [Photon] TCPClient unstable in 0.4.6

mhaack, 2015-10-06:
See https://community.particle.io/t/tcpclient-unstable-in-0-4-6 for discussions.

FW 0.4.6 has introduces some wired, frequent crashes (red flashing SOS followed by one red flash) when using the TCPClient.

I noticed this because two different application using the SparkFun Phant library crash now often on 0.4.6. Before on 0.4.5 the run stable for days. The issue can be reproduced using the SparkFun Phant example in the Web IDE.

*Steps to reproduce :*

* Load the SPARKFUN-SPARK-PHANT example in the WebIDE
* Change line 32 of TESTSTREAM.INO to a lower value, for example 5000
* Flash the Photon
* Wait

From my experience the time till crash varies greatly. I saw crashes every 2 minutes but also stable app. > 1 hour.

pkourany, 2015-10-07:
I have an app that fetches weather data for display on a Nextion HMI unit and it worked flawlessly with 0.4.5.  With 0.4.6 the Photon gets into SOS regularly though not always consistently as the OP stated.

virtualman18, 2015-10-09:
I am also facing the same issue where i am reading data from a serial port and posting the data to my servers API and i can usually get the first message through then the Photon resets itself. I've reflashed my Photon back to the the 0.4.5 firmware and have had no issues. Another thing to note is when disabling the API code and writing to the cloud directly it would sometimes disconnect and reconnect without an SOS but did not experience this issue on 0.4.5.

m-mcgowan, 2015-10-09:
There is a fix for this in develop - if you guys could test that would be much appreciated, since I don't have a test case for this. If anyone could share a minimal app that displays the problem that would be great to add to our test suite. 

When tcp was unstable in 0.4.4 it was because of the callbacks that are registered with WICED. These were removed in 0.4.5, and re-introduced in 0.4.6 (with some code that should have avoided any issues, but this clearly isn't behaving as expected.) For 0.4.7 we will remove the callbacks again.

closed 2015-10-20 by m-mcgowan