particle-iot/device-os #673: RGB.onChange with delay(1) and stalled RGB-LED

ScruffR, 2015-10-06:
I know it's not a good idea to call delay in an ISR or the `RGB.onChange()` handler, but I did it anyway - just for fun (actually to test `delay()` vs. `Particle.process()` to service `Particle.subscribe()` and `Particle.function()`).

And so I found that without `SYSTEM_THREAD(ENABLED)` even a call of `delay(1) brings the Photon down to a SOS hard fault (1x red).
Additionally a blocking handler also stalls the onboard RGB-LED (even with multithreading on).

```
#include "InternetButton/InternetButton.h"

//SYSTEM_THREAD(ENABLED)

InternetButton b = InternetButton();

void setup()
{
  b.begin();
  
  RGB.onChange(doRing);
}

void doRing(uint8_t R, uint8_t G, uint8_t B)
{
  static uint32_t ms;
  static int l = 1;
  
  if (b.buttonOn(1) || b.buttonOn(2) || b.buttonOn(3) || b.buttonOn(4))
  {
    l %= 11;
    l++;
    b.allLedsOff();
    b.ledOn(l, 50, 0, 0);
  }

  if (b.buttonOn(1))       // without multithreading
      delay(1);            //   this crashes immediately 
                           // with MF it survives
      
  if (b.buttonOn(2))
    while(b.buttonOn(2));  // this stalls onboard LED
    
  if (b.buttonOn(3))
    while(b.buttonOn(3))   // this one too
      Particle.process();
}
```

m-mcgowan, 2015-10-06:
As you mentioned in the start, the RGB.onChange() handler is called from ISR or from whatever context is updating the LED, which can also be the system thread, so blocking calls are strictly off limits. ;-)

As with all ISRs, delaying is not permitted. The fix here is to be as clear about this as we can in the docs, and suggest an alternative where the RGB values simply saved and picked up by another thread or the main loop if the user code really wants to block doing things not permitted on an ISR.

technobly, 2015-10-06:
onChange() is called every 1ms, so any realtime activity greater than 1ms in the handler will quickly create a stack overflow.

We kind of cheat in the example code and put the analogWrite(r_pin, r); etc.. calls in the handler, to keep the code simple to understand.  Ideally these calls should be made in loop() as Mat suggests.

technobly, 2015-10-06:
Also, thanks for being so diligent testing with testing @ScruffR!  It really helps :)

ScruffR, 2015-10-06:
Understood - so the best thing is to stress that `onChange()` actually is called via or similar to an ISR.
That should prevent people from doing what I did ;-)

Still interesting that `SYSTEM_THREAD(ENABLED)` helps to prevent the SOS hard fault (even with something like `delay(200)`)

m-mcgowan, 2016-03-14:
As a fix we can add a check in `delay()` if this is being called from an interrupt (we have `HAL_IsISR()`) and if so, to no-op the delay.

m-mcgowan, 2016-03-18:
#786 touches on this

m-mcgowan, 2016-04-08:
To sum up:

- delay() will check the HAL_IsISR() flag, and if true, change it's behavior to be a simple blocking delay rather than run any background processing.

PR #959 (2016-04-27): HAL_IsISR
adds HAL_IsISR() and uses this to skip calling the background loop from delay(). fixes #673

---

Doneness:

- [x] Contributor has signed CLA
- [x] Problem and Solution clearly stated
- [x] Code peer reviewed
- [ ] API tests compiled
- [ ] Run unit/integration/application tests on device
- [ ] Add documentation
- [ ] Add to CHANGELOG.md after merging (add links to docs and issues)
commit b43f526e (2016-04-27): adds HAL_IsISR() and uses this to skip calling the background loop from delay(). fixes #673
commit c39ab50c (2016-08-01): adds HAL_IsISR() and uses this to skip calling the background loop from delay(). fixes #673
commit d3eed6f6 (2016-08-01): adds HAL_IsISR() and uses this to skip calling the background loop from delay(). fixes #673
closed 2016-04-27 by m-mcgowan