particle-iot/device-os #677: [Photon] Particle.function()

pomplesiegel, 2015-10-07:


technobly, 2015-10-07:
Related to issue: https://github.com/spark/firmware/issues/659

As a work around, try letting loop() end more frequently to update function handlers.

pomplesiegel, 2015-10-07:
@technobly, sorry i accidentally filed the issue before filling in the body!    
How do you make loop() "end more frequently" than to have it encompass only `Particle.process()`, as the above code does? This should have loop() begin and end about every millisecond, correct?

technobly, 2015-10-07:
Hmm, that's interesting that the delay of only 1ms has such an impact, but I'm also guessing there is some interaction with the fact that it's running as MANUAL mode.  In general, the issue will be solved by including the pumping of function and variable handlers from Particle.process() when threading is enabled... but this is an edge case to look into.  cc: @m-mcgowan 

For your work around, please try this.  I'd be interested to know if the delay(1) is still required or not.
```c++
#include "application.h"

SYSTEM_MODE(SEMI_AUTOMATIC);
SYSTEM_THREAD(ENABLED);

int MyFunction(String command)
{
  Serial.println("my function was called!");
  return 1;
}

void setup()
{
  Serial.begin(9600);

  Particle.function("MyFunction", MyFunction);

  WiFi.on();
  WiFi.connect();
  Particle.connect();

  Serial.println("Example program");
}

void loop()
{
}
```

pomplesiegel, 2015-10-07:
Just tried it and the code you posted above using `SYSTEM_MODE(SEMI_AUTOMATIC)` works reliably.

So, to sum this up,
the issue occurs when using
````cpp
SYSTEM_MODE(MANUAL);
SYSTEM_THREAD(ENABLED);
````
together, and calling
````cpp
void loop()
{
  Particle.process()
}
````
without an additional delay within loop().

This was discovered because programs which sometimes are doing very little within loop() will not allow consistent successful function calls OTA.

technobly, 2015-10-07:
:+1: thanks for testing that

pomplesiegel, 2015-10-07:
Just realized that this is affecting our very full-featured program as well, as at certain times there is very little to be done within loop(). During that time, using FW 0.4.6 and `SYSTEM_THREAD(ENABLED)` we are sometimes not able to call any of our functions OTA.

So it appears this is more than an edge case, and rather a common symptom of using
````cpp
SYSTEM_MODE(MANUAL);
SYSTEM_THREAD(ENABLED);
````
and can affect any app

indraastra, 2015-10-09:
I'm also running into some issues related to this couplet of code that I've documented here:

https://community.particle.io/t/network-status-calls-and-system-thread-issues/16369/2

The end result is that networking and/or cloud connectivity seem to be compromised, probably due to synchronization issues.

As for this particular snippet of code posted by @pomplesiegel, I just want to point out [something I noticed in the docs](https://docs.particle.io/reference/firmware/photon/#system-threading-behavior) earlier today that applies to SYSTEM_THREAD(ENABLED):

```In MANUAL mode there is no need to call Particle.process() (but calling Particle.process() does no harm if you want to keep the code in place.) The system thread takes care of calling Particle.process() itself.```

pomplesiegel, 2015-10-09:
@indraastra, thanks for that! I totally missed that section of the docs. I'll try again with `Particle.process()` removed to see if there's a change in behavior.

pomplesiegel, 2015-10-09:
To reiterate @indraastra's point on the community, nice job on the multi-threading @m-mcgowan! Pretty amazing that our exact same code is running happily using the existing dev API. Our goal is simply to highlight any of these issues as early as possible, so they can be resolved before deploying a multi-threaded FW to our customers.

m-mcgowan, 2015-10-20:
I just tried this code and it's now working on the release/0.4.7 branch. The fix was to make sure the application and system threads have the same priority, so they are time-sliced when both busy waiting. (Previously the application thread had a higher priority, so the system thread was run only when the app thread was suspended, such as from a delay().)

closed 2015-10-20 by m-mcgowan

pomplesiegel, 2015-10-20:
Great!