particle-iot/device-os #680: WICED HTTP Server: Incorrect TCP Packet Handling

msolters, 2015-10-08:
It has been shown that the problem with SoftAP failure (most notably on Safari, some other browsers in iOS) is due to the fact that some some HTTP requests from the browser will naturally be fragmented into multiple packets.

**The current implementation of the WICED HTTP server only considers the first packet**.  This may work most of the time, but in the case of fragmentation, it will return `{r: 0}` to the first packet (spuriously representing "success" to the client), and then ignore the second or subsequent packet(s) *which actually contains the SSID & PW payload.*

This is a problem with the WICED HTTP server.  HTTP is a stream implementation of TCP.  HTTP does not care about packets.  It cares about the bytes in the stream.  The fact the WICED server does not actually accumulate packets and then defer processing the finished stream data means, quite simply, it is a very broken HTTP server.

Community user indraastra reports in more detail:
http://community.particle.io/t/softap-configure-ap-requests-thwarted-by-wiced-http-server/16331

And in addition it is discussed in the end of the great community saga, "Photon SoftAP Usage":
http://community.particle.io/t/photon-softap-usage/11035/178

msolters, 2015-10-08:
Here is a reference for determining an answer to the question:  How do I know I've reached the end of the HTTP request?
http://greenbytes.de/tech/webdav/rfc7230.html#message.body.length

indraastra, 2015-10-09:
(copying over from the community thread)

I just checked the HTTP server daemon in WICED SDK 3.3.1 and it seems to suffer from the same problem of not really handling HTTP as a stream. Despite it being a blocker for us, I'm not sure it'd be a worthy investment on your part to fix their broken HTTP server. In the interim, supporting a GET version of the configure-ap and connect-ap HTTP calls is a really great idea.

emilyrose, 2015-10-14:
@msolters, great sleuthing work! I believe I've encountered this bug myself. Thank you!

indraastra, 2015-10-14:
Python 3 script to trigger this issue:

https://gist.github.com/indraastra/6ba088ef4f7d5b1fb5af

m-mcgowan, 2015-10-14:
Thanks so much for the script, that was very useful. 

Rather than rewriting the HTTP server, I've added the minimum necessary changes to the HTTP callback interface so that we can retrieve the additional packets as needed.  It's in the PR above - those that can please try and let me know how it functions both for Chrome and Safari. Thanks!

m-mcgowan, 2015-10-15:
Here's the system-part2.bin file. Github doesn't support `.bin` as an extension, so I added `.txt`. 
[system-part2.bin.txt](https://github.com/spark/firmware/files/14266/system-part2.bin.txt)

To flash to your device, either use 

```
particle flash <device-id> system-part2.bin
```

or

```
dfu-util -d 2b04:d006 -a 0 -s 0x8060000 -D system-part2.bin
```

kmtm, 2015-10-16:
Unfortunately I tried both of the .bin methods and neither has worked via iOS or Mac OS X Safari (via Meteor).

msolters, 2015-10-16:
The Meteor app is using softap-setup v3.0.2, which should guarantee the correct HTTP verb.

What would be most helpful is if you could post:

1.  Any console errors?
2.  What does the network log look like?

msolters, 2015-10-16:
In other words, we are already taking into consideration the issue you mention. Perhaps it is not enough.

chees, 2015-10-16:
I'm using "softap-setup": "3.0.2" but the SAP.configure method still does an OPTIONS request which then hangs forever on Safari.
It's an OPTIONS call to 192.168.0.1/configure-ap

msolters, 2015-10-16:
Gotcha.  So it's clear that the `Content-Type` is insufficient to correct the request type.  @chees can you try a quick test for me?  Try using this pre-browserified JS:  https://github.com/msolters/softap-setup-js/tree/form-urlencoded/bin

Make sure to remove the NPM dependency on 3.0.2 official.  Including that JS file will let you create a `new SoftAPSetup()` as long as its available in the client.  Let me know if it changes the *type* of request that `configure-ap` sends.

indraastra, 2015-10-16:
@msolters: For whatever reason, the softap-setup-js fix didn't seem to make it in the js sent by the server.

indraastra, 2015-10-16:
Quick update: I can confirm that this is an issue even in my local copy of the meteor app. Running meteor seems to do something to softap-setup-js but the browserified code still contains the old version of __httpRequest:
```
$ meteor
....
npm-container: updating npm dependencies -- open, softap-setup...
```

```
$ grep -A 22 __httpRequest ./.meteor/local/build/programs/web.browser/app/client/softap.browserify.js
SoftAP.prototype.__httpRequest = function __httpRequest(cmd, data, error) {
   ...
	if((cmd.body) && typeof cmd.body === 'object') {                      // 241
		payload = JSON.stringify(cmd.body);                                  // 242
		opts.headers = { 'Content-Length': payload.length };                 // 243
		opts.method = 'POST';                                                // 244
	}
```

m-mcgowan, 2015-10-16:
Hi guys, can we mark this issue as closed, with the WICED packet handling added?

msolters, 2015-10-16:
Yes we should move this to spark/softap-setup-js

closed 2015-10-16 by msolters

m-mcgowan, 2015-10-16:
Many thanks! Great to hear this is fixed by the firmware patch.

msolters, 2015-10-16:
Let's hope this is all we need!

chees, 2015-10-16:
FYI I got the issue with the softap still doing an OPTIONS call working by deleting my .meteor/local directory. After that browserify seemed to pick up the new version which does work indeed :)
@indraastra That should probably work for you as well.

msolters, 2015-10-16:
Yeah we actually discussed & resolved this, see https://github.com/msolters/softap-setup-meteor/issues/3.

It's also made its way to http://photonsoftap.meteor.com