particle-iot/device-os #682: Clearing TX buffer in Serial

talybin, 2015-10-10:
I got a problem transmit over Serial1. The first char in first message is always garbage. Looking in HAL_USART_Init (https://github.com/spark/firmware/blob/develop/hal/src/core/usart_hal.c#L120) rx_buffer clears twice and there is no clear for tx_buffer which can be the reason for my problem. Suppose it is a copy-paste mistake. Please fix!

technobly, 2015-10-10:
Nice catch, this infamous copypasta has it's own restaurant chain:

https://github.com/spark/firmware/blob/develop/hal/src/core/usart_hal.c#L215

https://github.com/spark/firmware/blob/develop/hal/src/stm32f2xx/usart_hal.c#L263-L264

https://github.com/spark/firmware/blob/develop/hal/src/stm32f2xx/usart_hal.c#L263-L264

zerous01, 2015-10-18:
Mora pasta in

https://github.com/spark/firmware/blob/develop/hal/src/stm32f2xx/usart_hal.c#L148-L149

m-mcgowan, 2015-10-18:
grazie! non vogliamo la pasta!

technobly, 2015-10-19:
Oops, I see I pasted the same link twice up there instead of this one ^ which I'm sure I found before.  Thanks @zerous01 :+1: :spaghetti:

m-mcgowan, 2015-10-21:
TX buffer cleared in 0.4.7.

closed 2015-10-21 by m-mcgowan