particle-iot/device-os #685: Listening mode issue in v0.4.6

kennethlimcp, 2015-10-12:
I get this double rate blinking blue but not sure why it happened. Also the cipher type prompt was missing when this happened: 

```
SSID: xxx
Security 0=unsecured, 1=WEP, 2=WPA, 3=WPA2: 2
Password: xxx
Thanks! Wait while I save those credentials...

Derp. Sorry, we couldn't save the credentials.
```

Normal:

```
SSID: xxx
Security 0=unsecured, 1=WEP, 2=WPA, 3=WPA2: 2
Security Cipher 1=AES, 2=TKIP, 3=AES+TKIP: 1
Password: xxx
Thanks! Wait while I save those credentials...

Awesome. Now we'll connect!

If you see a pulsing cyan light, your device
has connected to the Cloud and is ready to go!

If your LED flashes red or you encounter any other problems,
visit https://www.particle.io/support to debug.

    Particle <3 you!
```

pomplesiegel, 2015-10-12:
"In another case after credentials are properly saved, the Photon breathes white and user code starts running:"
@kennethlimcp, I can confirm I have seen this same behavior. We were planning to file an issue this week.

During this time the Photon is definitely disconnected ( `Particle.connected()` returning `false`, and unavailable OTA). Has the WiFi chip been turned off, as a white LED would indicate?

karlhenricksen, 2015-10-13:
@kennethlimcp I've been seeing this behavior (breathing white LED) as well after sending wifi creds with 0.4.6, occurs maybe 1 in 30 attempts, much less frequent than https://github.com/spark/firmware/issues/663 but from the same process. When you say "credentials are properly saved" I assume it requires a reset first? That's what I'm seeing.

I still haven't found consistent steps to reproduce this, just seems to happen occasionally when setting up wifi. 

Notes: 
1) Using SYSTEM_MODE(MANUAL), SYSTEM_THREAD(ENABLED)
2) I'm unable to do a "particle get" or any function calls once in this state
3) Able to reproduce with wide variety of wifi security settings / speeds / router types, but possibly easier to reproduce when making router settings changes..

m-mcgowan, 2016-01-22:
The fast blinking blue was a LED state management issue which was fixed on 0.4.9. It would happen when WiFi was on but not connected, and the device entered and exited listening mode - the LED wasn't correctly restored on exit from listening mode.  

Breathing white is expected in MANUAL mode unless there is a call to WiFi.connect() or Particle.connect().  Variables/Functions aren't available until the cloud has been connected. 

0.4.9 has many improvements related to system threading, which may address the issue.

closed 2016-02-09 by kennethlimcp