particle-iot/device-os #690: Inconsistent Photon LED behavior when trying to clear wifi credentials

karlhenricksen, 2015-10-13:
If the Photon is in blinking green LED mode (searching for wifi, eg. new router or pw), I'm seeing various behaviors that are not consistent. This is generally a mild issue in that a repeated attempt nearly always is successful, but wanted to point out in case there are underlying issues that might need fixing.  

I see the following types of behavior with a Photon v 0.4.6:

Type 1 (100% reproducible): 
1) ensure the Photon is blinking green by removing your router
2) hold right button between 6-10 sec and release

Results: 
LED goes back to blinking green, instead of listening mode (blinking blue)

Type 2 (occasional): 
1) ensure the Photon is blinking green by removing your router
2) hold right button between 10-15 sec to attempt to clear wifi creds

Results: 
Photon remains blinking green, fails to go into listening mode or clear wifi creds

Type 3 (occasional): 
1) ensure the Photon is blinking green by removing your router
2) hold right button between 10-15 sec to attempt to clear wifi creds

Results: 
Photon goes into listening mode after 3sec, but never resets creds. Instead, the LED will turn off entirely.

jakeypoo, 2015-10-15:
I have found when trying to connect to a WiFi network the Photon code blocks for about 20 seconds, so presumably the mode button isn't being sampled or interrupts are disabled.

You may need to hold down the mode button longer in this case. It's meant to be a user interface, so just hold the button until the blue LED comes up / does the rapid flashing to indicate creds have been cleared.

m-mcgowan, 2016-01-22:
Thank you for the clear instructions. I tested against https://github.com/spark/firmware/commit/bc7d59b65bb8e22714b8d0186c5023747b91a10f and couldn't reproduce any of the cases mentioned.

It may be that these issues have been resolved since 0.4.6 was released. 

Alternatively, it may be that your application is affecting the behavior? To rule this out as a potential cause, please flash Tinker, or start the device in Safe mode.

m-mcgowan, 2016-03-18:
Closing as unreproducible - happy to re-open if new information comes to light.

closed 2016-03-18 by m-mcgowan