particle-iot/device-os #692: Problems connecting to Photon's SoftAP - stuck at Obtaining IP...

mebrunet, 2015-10-14:
We've been actively working with roughly 30 Photons for the last few months. We are seeing a strange inconsistency in our ability to connect to Photons' WiFi access points when the devices are in listening mode. 

After use (multiple connections, sets of credential, firmware flashes... etc.), some of the devices seem to get into a state where we cannot connect to their Soft AP access point. i.e. when we put the device into listening mode and select its access point from the list of WiFi networks, the connection fails. The connection process may either hang on message "Obtaining IP..." (Android 5/6), or returns error "DHCP lookup failed" (Ubuntu 12.04, Windows 7).

*Steps taken that do not consistently solve the issue:*
- Clearing the credentials,
- Resetting the device,
- Full re-flash of the latest firmware (v0.4.5 or v0.4.6: tinker + core modules). 

*Steps taken that consistently solve the issue:*
- Connecting to another AP. Resetting the Photon, then immediately connecting to the Photon's AP (within 10 seconds).
- Flashing firmware version 0.4.3. 

We've seen about a dozen devices get into this state at one point or another.

m-mcgowan, 2015-10-14:
Thank you for this detailed report. When a device exhibits this behavior, could you please put it in DFU mode then take a copy of the device's memory:

```
dfu-util -d 2b04:d006 -a 0 -s 0x8000000:0x100000 -U soft_ap_dhcp_fail.bin
```

and post a link to where I can download the file. Thanks!

mebrunet, 2015-10-14:
Will do.

mebrunet, 2015-10-28:
Okay, this has become a serious issue for us. 

We're having a lot of trouble establishing a consistent connection to our Photons' hotspots. For e.g. I captured the binaries from a Photon that behaves as follows:

A) Plug it in - we've wiped all credentials so it goes straight into listening mode.
B) Wait at least 30 seconds.
C) Try to connect to it's hotspot. You get stuck at "Obtaining IP". (See screenshot)

The only way to establish a connection is to.
1) Connect phone to another network or "forget" the Photon's network.
2) Press reset on the Photon.
3) Connect to the Photon's hotspot within a couple seconds.

In some cases step 2 needs to be preceded by holding setup for 10 seconds (i.e. re-clear credentials).

(Screenshot attached, but Github won't let me upload the binaries - so here's a [link](https://drive.google.com/file/d/0B4V2vkt-SuU9bEF1dEJCTk5MRWM/view?usp=sharing) )

[image]

mebrunet, 2015-11-02:
To further clarify the issue, this is not a problem where the Photon fails to connect to a WiFi network with the credentials provided. This is a problem where a device (laptop or phone) fails to connect to the Photon's Access Point (in order to scan and provide credentials).

m-mcgowan, 2015-11-03:
Hi @mebrunet. Thanks again for the memory dump. I flashed this to a photon and followed the steps several times to try to reproduce the issue, but alas I couldn't get the device to fail in the way described.

I'm using the latest Particle App on Android 4.4.2. 

I'm still wondering if there are steps that I'm skipping or misinterpreting.  To be completely certain I am performing the steps as you are, how about we schedule some time together (remotely) to reproduce the problem in tandem? cc @jonlogan

mebrunet, 2015-11-04:
Okay sounds good. This is a big priority for us, so I can make myself quite
flexible. When are you available?

In the meantime, perhaps try connecting using HTTP softap a few times. I
can't comment on the use of the particle app. I experience the problem
while connecting to the Photons WiFi AP manually before running a
connection script. (Note the issue arises before the script is set to
execute, i.e. I am rejected from the Photon's network.)

Sent while mobile. Please excuse brevity and typos.

mebrunet, 2015-11-09:
@m-mcgowan any thoughts on timing for a call/skype? I would love to figure this out.

MORA99, 2015-11-27:
This happened to a photon of mine today, I was able to set the wifi credentials by setting my android phone to use static ip in the 192.168.0.x range and then use photonsoftap webpage, since the cell app didnt work (seems to connect to wifi using dhcp again)

While I was able to get it working, I doubt my average customer would, its a pretty big issue for launch.

m-mcgowan, 2015-11-27:
I had a call with mebrunet and his current theory is that the problem only appears when they mount the photon to their board, and some kind of wifi interference/attenuation is happening. 

@MORA99 - was your photon mounted to a board? are you able to get the photon to eventually reconnect after resetting? are any other steps you know of to help us reproduce this issue?

MORA99, 2015-11-28:
Yes, the photon is mounted using the SMD side pads to the edge of a standard FR-4 PCB.
I removed ground plane on both sides, so theres not a massive amount of copper beneath it.

I tried setting it up 5times or so and didnt get it to connect, but with static ip it works, and once that initial setup is done it seems to work ok.

When I returned home today and wanted to reconnect it to my private wifi, the same issue happened again, and this time static ip solved it too.

mebrunet, 2015-11-28:
Have you seen the issue when the photon isn't mounted?

We've been testing on about 50 devices and found the following:

- We're unable to recreate the issue when the Photon is unmounted.

- The issue occurs more on our 1st gen PCB (with ground plane) than on our
second gen PCB (where we removed the ground plane around Photon's antenna).
Note we're also using a Photon with no headers, connected via side pads.

- If we use a lower amperage power supply the problem get worse.

- if we try to connect to the hotspot with the Photon at a great distance
(e.g. in the next room) the problem gets worse.

- if we use an external antenna the problem basically goes away (I.e. only
recreated at a great distance).

So our current hypothesis is that somehow the network forming portion of
the SoftAP process is more power intensive (or buggy), and the signal
strength dips during a few key calls, causing the process to fail.

I should add that on some devices we've seen connection failure (i.e.
losing connection to the hotspot) durring subsequent HTTP calls, but it is
much less common than failure during the first connection.

Sent while mobile. Please excuse brevity and typos.

MORA99, 2015-11-28:
I havent seen it with unmounted photons, but my sample size is 6 sofar, so not alot.
Do you think moving to a header mounted photon and/or removing the PCB directly below the photon as much possible solves the issue ?

peterqu, 2015-11-29:
I took a quick look at the PCB.  It looks like the power trace are very thin (10mil?).  Those traces probably does not provide sufficient current to meet peak power requirement.  This would explain why you are seeing intermittent issue with WIFI, as that is driving the peak power.  This also make sense why you don't see the problem with unmounted Photon, as the Photon would be powered through a USB cable which has a higher current carrying capacity than a 10mil copper trace.

Maybe you need a PCB engineer to look at your layout?  Maybe @jonlogan can help.

MORA99, 2015-11-29:
The VIN trace is 16mil, guess I should route it more carefully and use a thicker trace.

The GND plane is on both sides but iwth 12mil connectors, of which theres is only 1 on each of the 2 GND pads, in hindsight it would have been better to make a thick trace from the gnd plane to the pads.

Thanks for the suggestions, I am planning to make rev2 using headers however, so it wont be directly comparable.

peterqu, 2015-11-29:
Hello,

Why make rev2 with headers?  Usually headers is not preferred, as it adds additional cost of the header, a manual assembly step of placing through hole header and higher contact resistance.  I think you should try to figure out the root cause in your PCB design.

Are you using 1Oz or 0.5Oz copper? How long is your VIN trace? A 16mil 3-inch long copper trace on a 0.5Oz copper PCB board has a resistance of 180mOhm, probably too high for a power trace.  Also, do you have any ground loop? motor driver? high speed circuit on your board?  All those might contribute to electrical noise on the board.

MORA99, 2015-11-29:
Partly to allow easy replacement of the photon if it fails, desoldering it has the risk of damaging the board, but maybe its not that good an idea to use headers, I will keep your advice in mind.

The VCC trace is 338mm with 1oz, so around 400mohm resistance.
The onboard regulator provides 5.1volts, the photon datasheet says "As an input, supply 3.6 to 5.5VDC to power the Photon".
From the power table the largest current draw listed is short bursts of 430mA (Although a general text advises using a 1A power source).
Even 1A at 12mil should not fall below 3.6V.

Just to clarify the board works fine once its running, but in both test locations, its within 5-20meters of a access point, so it may well be using less current than in AP mode.

Theres no circuit on the board that is not controlled by the photon, so when its in AP mode, it should be a pretty quiet place.
The board is little more than a terminal breakout only active component besides the photon is power supply (TPS54231) and a DHT22.

peterqu, 2015-11-29:
Hello MORA99

Ease of replacement is definitely a good reason to go with header.  I want to share a tip on resoldering the SMD part.  You can buy a SMD removal kit from ChipQuik.  It has a Bismuth Tin solder that makes removal of SMD extremely easy.

Your calculation on the voltage drop is very good.  The only thing I would change is to consider the transient response of TPS54231.  Essentially when the load current suddenly increases, the output voltage of the regulator will momentarily drop and takes sometime to recover.  
[image]

A additional caveat about TPS54231 regulator is that is a switching regulator, so you want to follow the TI's recommended layout to minimize interference.  Otherwise you would get noise at multiple of 570KHz (570KHz, 1.14GHz, 2.28GHz, etc).  

I think for next step, we can try to isolate the issue to PCB or Photon.  If you have access to a Particle shield, such as the Programmer Shield.  Can you try the same setup on a Photon on a shield.  Do you see the same problem?

Thanks
Peter

MORA99, 2015-11-29:
I dont have any official boards, I can try to solder a photon to a bare board though to eliminate the regulator ?
I will gladly accept that the issue is with my board, but ideally there should be design documents that more clearly state the requirements, so that we can design for it without guessing.

Just tried to power it using 5v and that didnt work either, but the regulator may be putting a load on the line since its not powered.
Current draw was in the 100ma ballpark.

The 54231 is my standard power circuit, I have followed the recommended layout and the same layout is in use on a decent amount of boards with no issues, although the photon acording to datasheet is drawing 100-200mA more than I usually use.

If I am reading that figure correct, that shows a current draw change of about 1A and a resulting voltage drop of 200mV very breifly and then 100mV for about 100us ? thats at least 2-3times the current changes of the photon.
The output of the TPS is fed to the vin of the photon which feeds a regulator if im not mistaken, which presumably have caps and filters to deal with the USB power supplies out there.

peterqu, 2015-11-29:
Hello,

Solder a photon to a bare board is a great idea.  You can skip the regulator and power the Photon directly through the USB.  Let us know if that fixes the issue.

Once we can confirm the issue is somewhere on the board.  I can share with you a few tips to improve the board design for your next rev.  

Photon indeed has on-board regulator and decoupling capacitor that would reject noise on the supply.  In general, I found that Photon is pretty forgiving in handling input fluctuations but it has its limit.  You can help to improve the supply voltage by using parallel decoupling capacitors, better capacitor component,  etc.

I also agree that documentation has room to improve.  The challenge is to address every nuances in electrical engineering while keeping the document concise.  Since Photon shield is open source, you can fork an existing shield as an starting point to minimize some risk in PCB design.

mebrunet, 2015-11-29:
I can pipe in and say in both our PCB designs where we were experiencing
this issue, we were powering the Photon directly via the module's USB jack.
Our boards were extremely simple, and we were not drawing power for
anything else during the first connection process.

It really seems like the issue came from WiFi signal attenuation do to
interactions with the board's ground plane and the box enclosure...

Sent while mobile. Please excuse brevity and typos.

peterqu, 2015-11-29:
hi @mebrunet

If possible, Can you also test your photon on an official Particle shield?  Do you observe the same problem?

Thanks

mebrunet, 2015-11-30:
Hi @peterqu

I presume you're asking that to rule out a QC issue with an individual
Photon? If so, we've got 100 Photon's here, about 50 of which are mounted
to PCBs or in products for testing. So I can confirm it's not a problem
with a single module.

I recently requested half a dozen modules from another batch to verify it
wasn't a larger scale QC issue, however I haven't had a chance to test them
yet.

At this point we're just very relieved that using an external antenna seems
to resolve the problem. (There were a couple embarrassing demos, and
unhappy early customers... haha) I'll keep you posted as we figure out more.

Happy to test in an official shield if you still deem it useful. I could
also arrange to send you one of our products if you'd like to test it
directly.

Cheers,

peterqu, 2015-11-30:
Hi @mebrunet 

Sorry to hear about the embrassing demo and unhappy earlier customer. An IoT device requires engineering support across multiple disciplines: electronic, firmware and software. That is why I want to quickly triage this issue into a PCB or Firmware problem so you can quickly get the right help.

I have seen multiple cases where issues in PCB show up as instability or quirky behavior in firmware, especially in WIFI-related function.  After adopting good PCB layout practice and improve the PCB design, the firmware runs stable and without any error.

I am suspecting that there is some issue with your custom PCB.  By testing the Photon on a known-to-be-good PCB such as one of Particle's Sheilds, we can eliminate the effect on your custom PCB while keep other environmental variables the same.  If the issues goes away with the Particle Shield, then we know to focus our efforts to improve your PCB.  You can purchase a shield from either Particle or Particle distributor such as Sparkfun.

I am not closely affiliated with the Particle, so I don't know too much the specifics about the Photon production nor QC.  Based on my experience in electronic in general, I don't think quality issue of Photon is an likely cause.  The Photon PCB are impedance controlled and have much stricter manufacturing tolerance than your typical PCB.  All the WIFI-related components were sourced from manufactures that have ISO 9001 or similar Quality System (No ESP8266 here, haha ;-)).  In addition, the behavior your have observed requires manual intervention and would probably fail the production automated testing. 

If you can do the test with a Particle Shield and report back the result, that would be great.  We want to help you fix any issue and ship a quality product.

mebrunet, 2015-11-30:
@peterqu

Embarrassing demos are part of the game. Nothing we didn't expect. :) And to be clear, I'm in no way trying to blame Particle for them.

I'm happy to see if we can recreate the problem with an official Shield. Can you point me to a shield that accepts the Photon soldered on as an SMD component? Mounting the Photon with headers feels pointless (given our current hypothesis and the existing data on the problem), and all the shields I've seen for sale so far come mounted with female headers. 

You suggested that @MORA99 solder the Photon to a bare board and power via PCB. Our board is pretty close to "bare", (as mentioned, we're using the Photon's built in USB jack for power). But I think a valuable new piece of information would be to solder one up without the other components and see if we can recreate the issue.    

Also, can you clarify your relationship with Particle, just so I understand who I'm speaking with.

Cheers,
M-E

jonlogan, 2015-11-30:
@mebrunet allow me to chime in here --  Peter is a member of Particle's (relatively new) professional services team.  He's a very senior full-stack consulting engineer who helps Particle customers with targeted EE, FW, and SW tasks.  For this particular topic, he saw a chance to provide some handy guidance on troubleshooting.  @peterqu thanks for moving the conversation forward! 

Our internal platform team doesn't have the capacity to provide dedicated review services for our customers' end-products, which is a key reason we built up a scrappy services team.  Generally speaking, Peter is available to engage with customers for design reviews, or short/medium term engineering projects (on a time and materials basis).  

I'm the head of customer success at Particle, and Peter works on my team.  If you're interested in PCB review services or engineering support, let me or Peter know!  

Thanks, 
Jon

peterqu, 2015-11-30:
Hello @mebrunet 

I would start with the Programmer Shield for initial testing. 

There should be very little difference between Photon with header vs Photon surfaced mounted via the castellated edge.  Direct soldering onto the castellated SMD edge actually will likely to create a better contact with lower contact resistance.  The main difference between SMD mount vs header mount Photon is that in SMD mounted Photon, there is a FR4 dielectric directly underneath the Photon.  FR4 dielectric has high permeability for electromagnetic wave and it is expected to be "transparent" for the WIFI signal.  Not to mention the WIFI signal emerges from the antenna side, not the underside where it is facing the PCB FR4 dielectric.  So, I expect on a good carrier PCB, surface mount Photon vs header mount Photon should behave the same way.

In @MORA99  case, his PCB needs to improve in a few areas:
1) Increase power trace thickness to meet peak power requirement.
2) Switching regular has fundamental frequency of 0.57GHz and 4th harmonic at 2.26GHz.  The 4th harmonic is very close to WIFI's 2.4GHz.  Electromagnetic issue from the switching regulator might affect WIFI performance.
3) Copper traces over area without copper ground plane.  If those traces are high speed, it will create ground loop and cause electromagnetic emission.
In my quick look of @MORA99's PCB, the power and regulator seems to be the likely issue.  That is why I recommend him to solder it onto a bareboard and power via USB.  This will bypass the the issues with power trace and regulator.

@mebrunet : Your PCB is different from @MORA99 , so it might have different set of issues.  We have to review it separately.

mebrunet, 2015-11-30:
Okay - thanks for all the clarification.

Just so you know, for the last week or so our team has been thinking about the problem as a hardware design/compatibility issue, and not a problem in firmware. (Albeit we never saw this issue while still using system firmware v0.4.3, I had fewer devices assembled at that point).

The issue we are experiencing is quite intermittent. The same PCB/Photon may work fine through a dozen connection procedures before it displays a problem. We had a ground plane directly under the Photon (which I presume would not appear as "transparent"), we have since cut it away around the antenna, and seen a great improvement. Adding the external antenna completely solved the problem for us (at least in testing so far). 

Note: your design guideline "ground plane and signal keep out recommended on all layers" can be misinterpreted as "there should be a ground plane and no signals here". As opposed to what we now understand to mean "there should neither be a ground plane or any signals here". Consider rewording.

For us, this issue is pretty much closed. We haven't seen the problem when the Photon is unmounted, or in a breadboard, and we've found a solution in our PCB / hardware design. I really don't expect we'll see this issue when the Photon is mounted via headers to the programmer's shield. But I may put some time in for curiosity sake.

peterqu, 2015-11-30:
Hello @mebrunet 

I am glad that your team has figured out the the issue.  

I want to reiterate your point and make it educational for the community.  When you have any copper (traces, ground plane, power plane) below antenna keep-out area, the copper would form a "Faraday Cage" and block electromagnetic waves such as WIFI signal.  

Thank you for the suggestion to change the wording. The word "keep-out" is a standard term used by PCB engineer to describe any area where you don't want something to be in, copper keep-out, silkscreen keep-out, etc.  In case of Photon, it means there all the copper should be keep outside of the keep-out area around the antenna.  I can definitely see this wording can be confusing and I will update the documentation.  

As for the header vs surface mount Photon, you may find surface mount Photon to be necessary for larger scale production.  Photon with header requires an extra placement step and your assembly house may charge you extra for that.  By the time you are ready to make a millions of your gadget, you would have perfected the PCB design and should have no problem with photon mounted either way.

mebrunet, 2015-11-30:
Haha, "keep-out" is pretty self explanatory, it was the grammar that got us.  
We have been, and will continue to use surface mounted Photons for that reason :)
Thanks for your input.

MORA99, 2015-11-30:
Okay, I have done a bare board test, and both with USB power and 5V via pads just below the photon it works.

But seeing how it works every time if I use a static IP, and it works fine in regular mode, it must be borderline acceptable.
I have a few traces below the photon, but in safemode+listening they would be idle, whereas during operation they would be in service.

peterqu, 2015-11-30:
@MORA99 

I am glad it works.  Looks like the power trace and/or switching regulator is the problem.  For you next revision of PCB, I recommend:

1) Increase the power trace width to >40 mil.
2) Double the decoupling capacitor by parallel two caps.  
3) Add a option for a linear 5V regulator such as UA7805.  The linear regulator will not create switching noise.  Having option for both linear and switching regulator will help you debug whether you need to do more work to eliminate EMC from your switcher.  In the end, you want to use the switcher for its efficiency.
4) Avoid routing signals over area where you don't have a copper ground plane.

Give that a try.

MORA99, 2016-01-01:
Seeing how this only happens while in listening mode, is my user code still running at that point, otherwise theres no way I can have high speed signals active at that time.

My Rev1 board has been running the last month with no issues, its only the SoftAP thats problematic, but I got Rev2 of my board last week and at first glance it was better, I was able to setup the network config while soldered to the board and then load my firmware with no issues.
The new board uses 50mil traces and 4 additional relevant capacitors.

Completed the board with the remaining components and wanted to verify that I could still redo the wifi config and it failed ...

So I spent a bit of time debugging the issue, and oddly enough if I flash tinker on the device and then go to listening mode, I can connect to the SoftAP.
Also with reduced/test user firmware SoftAP works, but when I flash my complete firmware I cant or rarely can get into SoftAP.

So it seems the listening mode does not reset into a clean enviroment, and something my user code does makes the SoftAP job harder and more prone to fail.
I havent been able to isolate the issues yet, but it is definitely related to what user firmware the device is running, since both simple user firmware and tinker lets me get into SoftAP almost everytime.

m-mcgowan, 2016-01-22:
If you are using `SYSTEM_THREAD(ENABLED)` then your application will continue to run while in listening mode. Without threading, the application code is blocked during listening mode.  I hope that sheds some light on the problem!

sazp96, 2016-01-30:
Hi @MORA99, we are experiencing the same issue (even running the new FW 0.4.9. Were you able to identify the root cause? 

We are planning to do a small PCB production run. So we are trying to figure out if this is a software or hardware issue. 

Thanks!

MORA99, 2016-01-30:
@sazp96 What are you using for power supply ?

A scope of our power design showed a significant drop in voltage (about 300mV) during the softAP process, although there was still more than 1volt to the minimum vin of the photon, it seems this drop is enough to make the DHCP server fail.

So I would say the photon during softAP is sensitive to voltage drops beyond the official spec of 3.6v, I am doing a 3rd PCB revision with a different power supply design(Traco power module) that limits the drop after sudden change of power usage better than the current one, but with chinese new year it will be a few weeks before I know if that worked.

I still think this could have been solved in software, but the photon is in part based on "WICED" firmware, which is the parts that are handling the softAP mode, so sadly its not likely that a fix will be made.

sazp96, 2016-01-30:
@MORA99 thanks for the response. 

So here is what we are using:
- USB Micro-B - 5 Volts at 2 Amps
- Power traces width of 20 mil
- Copper weight 2 Oz
- C1 = 220 uf - 16 Volts
- C3, C5, C6 = 0.1 uf - 16 Volts
- Connected on JP1 is a 3.5" TFT Display (The display back-light is on during the softAP setup)
- We also have a LED strip connected to the circuit but it is not used during setup.

Should we make the power traces wider and/or change the capacitors?

[image]

MORA99, 2016-01-30:
Interesting, so you do not have any local voltage regulation before feeding the phoron ?

If you have access to an oscilloscope you could try and measure the drop, but 750ma at 20mil is around 100mV, and the usb adapter could be providing less than ideal regulation.

Previously in this thread I was recommended 50mil traces at 1oz, but I changed that and it didnt solve the issue, maybe due to other factors.

peterqu, 2016-01-31:
@sazp96 

There probably are transient voltage drop on your supply line during WIFI transmission.  You can try add 3 capacitors (ceramic 0.1uF, ceramic 10uF, electrolytic/tantalum 100uF) between supply and ground near the Photon.  Those capacitors should help.

sazp96, 2016-01-31:
@MORA99 No, we don't have a regulator before the Photon. Instead, as recommended in the particle community, we have to capacitors. A "big" one and a "small" one to capture the different types of noise in the power signal. 

Unfortunately I don't haven an oscilloscope, but it is about time I get one. I used my multimeter connected to the Photon's VIN/GND and the voltage oscilated between 5.11V and 5.12V. That should be enough voltage, right? Or maybe there was a drop too fast for the miltimeter to notice...

@peterqu thanks for the tip. I added to the Photon's VIN/GND a 100uF ceramic and a 100uf electrolytic capacitor in parallel using some cables soldered to the PCB, but it didn't work. Having said that, all those handmade connections and the cables might be doing more harm than good. I will need to try it on the new version of the PCB. 

I found a consistent work around to get the SoftAp setup to work every time. Here is the repro:

1. Power the Photon (in my example it is running Tinker)
2. Try to connect to the Photon AP. Wait for your phone to get stuck in "Obtaining IP"
3. Unplug the Photon and wait for 2 seconds
5. Plug the Photon again
6. You phone should change from "Obtaining IP" to "Connected"
7. Finish the SoftAP setup

@MORA99 and @mebrunet does this work around works for you?

@peterqu does this behavior share any light into what could be the root cause of the issue?

peterqu, 2016-01-31:
@sazp96  The leads of the capacitors need to be very short in order to minimize ESR and to work properly..    

I would suggest you try to following to isolate the problem to power.  Since you are using Photon, plug the USB power cable directly to the USB connector on Photon, do you observe the same behavior?  If you don't observe the issue, then you can pin point the power on your board and in the wall plug.

If you are going to re-spin the board, it might be worthy to talk to a board designer for advice.  If there are power issue, there are likely hidden electronic design issues that would show up later as weird software behavior or failed certification application.

MORA99, 2016-01-31:
@sazp96 Do you enter listening mode after replugging the photon? that would give at least a 10second delay with no wifi.

The way I can get the process to work, is if I connect manually to the wifi with a fixed ip in the 192.168.0.0/24 range and using the webbased setup http://photonsoftap.meteor.com/

If you use the photon app it will try to reconnect with dhcp and fail.

Sometimes I can get the softap process working as it should, but its pretty rare, so hard to determine what makes the difference when I get it working.

mebrunet, 2016-01-31:
We've completely solved this issue by taking the following measures.

1) We power the photon directly via its mounted usb jack. (It's actually with this that we power the whole board including another radio). (Still saw the problem).
2) We cleared a large area (no ground plane, no traces) around the Photon's antenna. As per the design specifications. (Solved the problem 95%).
3) We now use an external WiFi antenna (via the uFL connector), and haven't seen the problem since.

I wrote a React/jquery softAP setup process using the same RSA library I included in the softAP tools I open sourced. We've run off a couple hundred devices with no issues, and are now scaling to our first batch of 1000.

m-mcgowan, 2016-01-31:
Congratulations @mebrunet  - really glad to hear you found a solution!

sazp96, 2016-01-31:
@peterqu we have tried different USB power supplies and different USB cables without any luck. We have also started powering the board using the photon's on board USB jack but we keep seeing the same issue. 

@MORA99 to keep it consistent, all the photons we are testing have no credentials stored (so they enter listening mode as soon as they are powered on) they are running tinker and system FW 0..4.9. 

Your tip of setting a static IP works great. Unfortunately it will be too cumbersome to ask our customers to do that during setup. But it is great for our testing. 

@mebrunet congrats! I'm glad you guys figured it out. We have tested doing the 3 things you mention with no luck. 

Earlier in the thread you mention switching from using the SMD to using the Photon with headers. Which version are you using now? We don't see this problem with our old (and different) PCB boards that use the Photon with headers.

[image]

peterqu, 2016-01-31:
@sazp96 

 You have connected to copper at top of the antenna keep-out.  This forms a current loop and made your board into a huge loop antenna that absorb all the electromagnetic interference in the air and couple them to your board.  So it is no surprise that your are running into WIFI issue because the board design is high susceptible to electromagnetic interference.

 
[image]

sazp96, 2016-01-31:
That fixed it! I cut the PCB at the top to break the loop and the Photon has started connecting consistently. Thanks @peterqu!!

So, even though we didn't have any copper in the "keep-out" zone, the ground plan that was around the zone was creating a current loop that was generating electromagnetic interference, right? Is the best practice then not to have a ground closed circuit around the "keep-out zone"?

Solved:
[image]

peterqu, 2016-01-31:
@sazp96 

The ground loop is the first obvious problem I have found, and looks like it is one of the major problems.

I am glad that the problem is fixed.  When you have ANY copper that create a whole or even a partial loop, it became an antenna when the length of the copper exceeds 1/4 wavelength of the electromagnetic waves.  Unintended antenna will cause many issues and dropped connection is one of them.

sazp96, 2016-01-31:
@peterqu, @MORA99, and @mebrunet, thank you so much for all your help and feedback! 

You guys are awesome!

peterqu, 2016-01-31:
I have another thought, since people in the community often run into power and WIFI issues when creating their own PCB.  How much interest is out there for a reference board design that includes properly mounting of Photon/P1, power regulation and RF optimization?  If there is enough interest, I might create an reference design when I got some spare time.

sazp96, 2016-01-31:
That would be extremely useful. It would had saved us a lot of time.

mebrunet, 2016-02-01:
Ahh, this brings me back to my days of solving Maxwell's PDEs... Such
beautiful black magic all that non Kirchoff/Ohm simplified EMF.

A PCB template would be awesome. Especially if it included a P1 verison
with a power suply! Happy to contribute, what would you need?

sazp96, 2016-02-01:
Same over here. Let us know if there is anything we can do to help!

peterqu, 2016-02-01:
Yeah, certainly we don't have to solve Maxwell PDE to do this line of work.  Otherwise I would have quit EE in my undergrad.  But, there is a 200 pages tome on how to design a high speed board.  Ground loop is somewhere in the first 20 pages.

Talk about the reference design, which feature are useful to include?

MORA99, 2016-02-01:
Interesting, if the photon is affected by a ground plane just above the keep out, shouldnt the keep out be larger ?

The photons antenna in my 2nd rev is not at a edge of the PCB because orientation would hinder access to the USB connector then, so my antenna is in the "middle" of a large ground plane, only interrupted as per the keepout area.

I tried connecting an external antenna, does listening mode automatically use this? (recent firmwares disabled automatic antenna selection for user firmware)

sazp96, 2016-02-01:
To invite more people into the conversation, I have started a thread on the Particle Community to discuss best practices for PCB design related to ground loops and keep out zones.

http://community.particle.io/t/photon-smd-keepout-zones-and-ground-loops

m-mcgowan, 2016-03-18:
Thanks to everyone for sleuthing this issue! Great discussion and wonderful to see the discussion continuing in the community.  Since the issue appears to be caused by board design, can we close out?

HardWater, 2016-03-18:
Hi Mat, not to be a naysayer but when we were experiencing this problem we had it occur a number of times while working with Photons that were not in pcbs had an external antenna attached and were powered via a Particle usb cable. So though I am happy that the problem has seemed to go away for others by making pcb and antenna changes I feel there is still something unresolved here.

MORA99, 2016-03-18:
I dont think the original problem has changed, a product that requires specific PCB features should list these.
The photon is obviously very sensitive to PCB layout more than what the keepout leads to believe.

Many posts are shots in dark about poor power supply and thin traces, yet the problem seems more related to antenna/wifi signal than actually power issues.

sazp96, 2016-03-18:
I agree with @MORA99. It is really hard to figure out what extra requirements are needed in the PCB to make the Photon's SoftApp to work. 

To be more precise, this issue affects only the Photon SMD and not the Photon with headers. A possible solution could be to make the Keep Out zone of the Photon SMD bigger than the one for the one with headers. 

Just my 2 cents.

IanBoyanZhang, 2016-04-10:
Thanks guys! This is really informative and educational thread! Anyone has interest to put an directional antenna on PCB reference design?

[11 more replies]

closed 2020-02-28 by avtolstoy