particle-iot/device-os #697: LIN Bus Support

technobly, 2015-10-16:
Looks relatively straight forward to add LIN bus support to the Photon/P1/Electron (maybe Core... haven't looked yet):
https://community.particle.io/t/lin-bus-usart-flags-and-errors/16589/5

### Proposed API: 

`Serial.begin(BAUD, CONFIG);` // Where CONFIG = LIN_TX, LIN_RX10, or LIN_RX11 to enable/disable LIN master/slave mode (with 10 or 11 bit break) (can only be one at a time)
`void Serial.breakTx(void);` // To send a 13 bit break signal when in LIN master mode.
`bool Serial.breakRx(void);` // To check for a received break signal when in LIN slave mode.

Much work would still need to be done to create a functioning LIN Library, something a user could implement using these new Serial features.

### Example LIN Library:
https://github.com/gandrewstone/LIN/blob/master/lin.cpp

### LIN Reference:
https://en.wikipedia.org/wiki/Local_Interconnect_Network

m-mcgowan, 2015-10-16:
There are also these serial-related PRs

https://github.com/spark/firmware/pull/412
https://github.com/spark/firmware/pull/413

Does it make sense to try to encompass the goals of these PRs in a single API for USART configuration?

technobly, 2015-10-16:
Yes definitely!  I was considering those with the proposed API, which falls in line with the arduino API.  Didn't know someone already submitted a PR for it :)  Nice.

clay-to-n, 2015-10-23:
A USART config API would be great.  Opens up the Photon to integrate with a much larger range of other devices. :+1:

zakx, 2016-02-05:
Any news on this? Very interested in this too.

m-mcgowan, 2016-03-18:
@clay-to-n - following on from your PR on Serial Parity, can you see a way we can incorporate LN config on top of the existing API? Is it just a matter of defining new config constants?  

The HAL APIs need to be backwards compatible once released, so if there are changes we need to make to `HAL_USART_BeginConfig` to support LN, it would be great to have those done now before we release 0.5.x.

closed 2016-08-09 by technobly

ScruffR, 2017-03-10:
@technobly, after pinging you on slack, I tried also searching for _already closed_ issues regarding my SRF01 problem where I could not get half-duplex LIN_MASTER_13B work with that sensor.
And as it turns out, it does work, but the docs (including the prerelease-docs) fail to mention these functions

> void Serial.breakTx(void); // To send a 13 bit break signal when in LIN master mode.
> bool Serial.breakRx(void); // To check for a received break signal when in LIN slave mode.

But after I found this closed issue, I just tried and there it was - it does work after all.
Maybe the issue should not have been closed before this feature got "fully" documented.