particle-iot/device-os #698: i2c corruption due to task switching

pomplesiegel, 2015-10-16:
#### My assertion 
Photon (and possibly Core) i2c data corruption/error issues are caused by task switching during critical sections of i2c calls. More exaggerated failures, such as we observed in #666, are caused by`SYSTEM_THREAD(ENABLED)`, allowing more task-switching than ever before during a timing-sensitive i2c section of code.

#### Background
We rely on communication with a MCP23017 port expander over i2c for real-time interaction with a front panel. We perform about 20 back-to-back i2c transactions with this port expander every 10ms to ensure real-time performance with the front panel switches and LEDs. I2c issues arise easily and quickly using this setup, as we achieve about 1 million transactions every 8 minutes. Therefore, we are very sensitive to any changes in the `Wire` library. 

In addition, we track whenever a transaction has been unsuccessful ( `Wire.endTransmission() !=0` or `Wire.requestFrom() == 0` ), record this in serial logs and retry the i2c transaction. If the transaction is unsuccessful after 7 repeated attempts, we call Wire.reset() to free the bus, as this peripheral is likely locked up. 

We have always seen about 1-2 failed i2c transaction(s) per 30 seconds when communicating in real-time with this port expander, but the exchange can often recover by simply retrying the transmission. This was true for both Core and Photon. Adding `delayMicroseconds(1-50uS)` as recommended by [many forum posts] (https://community.particle.io/t/i2c-lcd-freezes-photon/15666/11) - and [again] (https://community.particle.io/t/using-an-i2c-spi-lcd-on-the-spark-core/2116/64) seemed to reduce the error rate, but never did so dramatically. But, with our retry scheme + Wire.reset(), the system was stable enough.

#### Today
After slowly eliminating other variables, we are now positive that`SYSTEM_THREAD(ENABLED)` causes much more severe i2c data corruption to occur, as our commands to the front panel were now being mangled, causing LEDs to erroneously turn on and off. With the exact same code running on system FW 0.4.6, but `SYSTEM_THREAD(ENABLED)` commented out, the issue disappears and our program functions normally. 
Basically,
* With Multi-threading, LED state corruption repeatedly occurs after about 5 test trials
* With Single-threading, corruption occurred 0 times after 250 identical test trials. 

#### Experimental changes
Following @m-mcgowan's suggestion that perhaps task switching was interrupting i2c in the middle of a critical section of code, I added protections around all critical `Wire` calls we make in the format below:
````cpp
os_thread_scheduling(false, NULL);
< call to HAL from some TwoWire method >
os_thread_scheduling(true, NULL);
````

For example,
````cpp
uint8_t TwoWire::endTransmission(uint8_t sendStop)
{
  os_thread_scheduling(false, NULL);
  uint8_t result = HAL_I2C_End_Transmission(_i2c, sendStop, NULL);
  os_thread_scheduling(true, NULL);
  return result;
}
````

##### My local copy of this modified TwoWire class [here] (https://gist.github.com/pomplesiegel/c8d2fb578d0af887c2a3)

#### Results
When compiling the tip of `develop` and our app (including `SYSTEM_THREAD(ENABLED)`) locally combined with the changes to spark_wiring_i2c in the [above gist] (https://gist.github.com/pomplesiegel/c8d2fb578d0af887c2a3), we have
* No front panel LED corruption (fixes issue seen with `SYSTEM_THREAD(ENABLED)`)   

but more importantly...

### No i2c minor errors/miscommunications at all 
even after > 3.75 hours of continuous real-time operation and logs. Remember, that's about 28 million i2c transactions with 0 errors with this port expander. 

Without these task-switching protections inserted, when running FW 0.4.6 or 0.4.5 in single-threaded mode, we were seeing an i2c error about every 30 seconds. This was consistent with performance on the Core as well, even going back to early 2015. 

In #666 @m-mcgowan mentioned
`I'm kind of surprised we haven't seen failures before since WICED is always task switching` 

**My naive assertion is that these i2c data corruption failures have always existed, and been caused all along by task switching during critical sections of i2c code.** We have always seen these failures, but those who exercised the i2c bus more thoroughly were more likely to see the issue. Perhaps this protection would fix the longstanding issues observed with this common port expander, for those who use it in a real-time application:
https://community.particle.io/t/using-an-i2c-spi-lcd-on-the-spark-core/2116/116
https://community.particle.io/t/i2c-lcd-freezes-photon/15666
https://community.particle.io/t/i2c-lcd-display-getting-corrupted-solved/9767?page=2
...

#### My Proposal
* Integrate these task switching protections around all `Wire` methods which communicate with the i2c controller. 
* Refine the code provided in [my gist] (https://gist.github.com/pomplesiegel/c8d2fb578d0af887c2a3), as with this code added **we have seen Photon system instability during resets via the onboard button**. I'm guessing `os_thread_scheduling(false, NULL);` is possibly too restrictive - less of a scalpel than methods others would likely use to protect against task switching during critical i2c moments.
* Investigate whether these same protections should be applied to `SPI` or other buses

This is all solely based on my observations and experiments using and modifying `Wire` within a full-featured real-time application. 

@m-mcgowan, what are your thoughts?

m-mcgowan, 2015-10-17:
This all makes perfect sense. Since you've done the hard work testing, I can get the Wiring changes in place for 0.4.7 to be released early next week. Thanks for your efforts in hunting down this issue!

I think the only remaining point to discuss is whether this has been an issue on the Core, since that doesn't use task switching, but there are still interrupts that may "steal" time away from timing-critical code.  If we feel that even with task switching disabled, that the remaining interrupts could also potentially cause issues, then we will have to go one level further and disable all interrupts, thus preventing both task switching and interrupts.

technobly, 2015-10-18:
@pomplesiegel I haven't addressed the task switching with a recent I2C fix I made, but I wanted to see if you could easily test this PR that makes some changes to the I2C library and make sure they don't adversely affect your I/O expander.  I've tested with the Sparkfun weather shield and no slave device connected, and those tests are working better than ever.
https://github.com/spark/firmware/pull/700

Thanks!!

pomplesiegel, 2015-10-19:
@m-mcgowan, that's great news! Happy to try out any i2c changes. We're not doing any further development on the Core, but I would be happy to QA any further changes. 

My only question is what would prevent the occasional system instability I saw on Photon when adding `os_thread_scheduling(false, NULL);` to the `Wire` methods. Is there a less restrictive call which won't cause crashes/delays to the system FW when pressing the reset button on Photon? I'd imagine that issue would show up elsewhere, for other interrupts? 

@technobly, I can certainly try out those changes tomorrow in conjunction with the task switching restricted code to make sure everything is working as planned. How does that sound?

Thanks!

technobly, 2015-10-19:
That sounds great @pomplesiegel, thank you!

pomplesiegel, 2015-10-19:
@technobly, just verified 3hrs of operation without any i2c errors. Looking good!

technobly, 2015-10-19:
@pomplesiegel great thank you for testing!  Just to double check, did you also have your `os_thread_scheduling(..., NULL);` calls in there as well or did you just disable system threading to test?

pomplesiegel, 2015-10-19:
Hi there, the os_thread_scheduling(...,NULL); calls were included as well.

technobly, 2015-10-19:
Ok cool, and it sounds like Mat is on board with those system threading changes for 0.4.7 so these should go together seamlessly.

pomplesiegel, 2015-10-19:
Fantastic! Thanks

commit a648ebaf (2015-10-20): Add critical sections around I2C operations for increased reliability. Fixes #698
closed 2015-10-20 by m-mcgowan