particle-iot/device-os #704: Photon: Possible issue with "unhandled" "brownout"?

ScruffR, 2015-10-20:
Recently there seem to pop up topics about Photons becoming unusable or loosing data due to power issues (e.g. battery drainage, cheap DC chargers, ...)

e.g.
 https://community.particle.io/t/photon-ran-low-on-battery-now-stuck-in-sos/16710
 https://community.particle.io/t/eeprom-persistence-issue/16514

Could this be helped via software or does it need hardware?
If the latter a proper warning in the docs for existing devices might be required.

nicodegunst, 2015-10-30:
I have experienced this also. It is a new and I believe related to 0.4.6 fw. I have worked with both the core and photon with all iterations of the firmware and have tested with the same battery-powered power source all the time, draining the batteries. So far it didn't happen that eeprom lost its data, and now suddenly it does.
I hope this will be resolved soon..

kikolobo, 2015-11-04:
I have similar issues where a photon running on battery power becomes unusable after the batteries are drained and then re charge.. The photon turns on either the D7 LED and has to be re flashed using JTAG... Or it looses wifi credentials after the reboot.

Don't know what causes but it doesn't happen with direct power ... Only when running on batteries and the batteries run low.

InMyOrbit, 2015-11-19:
I am also experiencing this issue. I used the board with the SparkFun battery shield and since then I have not been able to use the board continuously. I manage to configure the board using the cli-interface and perform a remote upload. But as soon as the device reboots it looses its wifi credentials and gets stuck. The RGB-led flashes in an infinite pattern like this: 25 seconds of cyan flash, 3 orange flashes.

Tried updating keys without success.

monkbroc, 2015-12-05:
[I proposed in the community forum to set the brown-out reset configuration registers.](http://community.particle.io/t/eeprom-persistence-issue/16514/39?u=jvanier) If you are having this issue please report back here if setting a higher brown-out reset level helps.

```
void setBrowoutResetLevel() {
  const uint8_t desiredBOR = OB_BOR_LEVEL3; // Supply voltage ranges from 2.70 to 3.60 V
  if(FLASH_OB_GetBOR() != desiredBOR) {
    /* Steps from http://www.st.com/web/en/resource/technical/document/programming_manual/CD00233952.pdf */
    /* See also https://github.com/spark/firmware/blob/aefb3342ed50314e502fc792f673af7a74f536f9/platform/MCU/STM32F2xx/STM32_StdPeriph_Driver/src/stm32f2xx_flash.c#L615 */

    /* To run any operation on this sector, the option lock bit (OPTLOCK) in the Flash option control register (FLASH_OPTCR) must be cleared. */
    FLASH_OB_Unlock();

    /* Modifying user option bytes */
    /* 1. Check that no Flash memory operation is ongoing by checking the BSY bit in the FLASH_SR register */
    FLASH_WaitForLastOperation();
    /* 2. Write the desired option value in the FLASH_OPTCR register */
    FLASH_OB_BORConfig(desiredBOR);

    /* 3. Set the option start bit (OPTSTRT) in the FLASH_OPTCR register */
    FLASH_OB_Launch();

    /* disable the FLASH option control register access (recommended to protect the Option Bytes against possible unwanted operations) */
    FLASH_OB_Lock();
  }
}

void setup() {
  setBrowoutResetLevel();
}
```

InMyOrbit, 2015-12-07:
This did not solve my issue. My device still keeps flashing cyan.

monkbroc, 2015-12-07:
@InMyOrbit I wouldn't expect the brownout level to fix the issue you describe. Hopefully when you do fix it the brownout level config will prevent it from happening again.

kikolobo, 2015-12-08:
I will try this solution tomorrow. I need to go to the field and reset the photon manually to be able to program it.

monkbroc, 2015-12-08:
@kikolobo You need to this only once since the setting is stored in flash, but it doesn't hurt to call it every boot in case you change Photon.

peterqu, 2015-12-08:
@monkbroc 
This is a great fix.  Though, I have not tested this on yet, I have seen multiple failure cases where this fix would be very helpful.  

The only thing I might be wary about is to put this code in the setup() and write to Flash with every system boot.   In STM32, the number of write cycles is 10k to 100k, and we expect the number of read cycles to be 100 times that of write cycles.  In some cases where the customer's carrier board for Photon has bad power supply or excessive noise from motors.  the POR and BOR will be triggered very often.  If the user firmware write to Flash on every startup/reset, the flash might degrade very quickly.

Maybe we should turn on this feature by default in the factory image?

monkbroc, 2015-12-08:
The function above only writes to flash if the brown out level had not been configured yet `FLASH_OB_GetBOR() != desiredBOR`

If it works well for those who test it here, it should definitely go in the system firmware.

peterqu, 2015-12-08:
@monkbroc 
Great, I didn't catch that part. Its excellent.

peterqu, 2015-12-08:
I have a variable power supply and I can do a power margin test to check the BOR function.  It would be great if you can give me a pointer to a test case, if you have already created one.

andyw-lala, 2015-12-08:
IMHO - this is a band-aid, possibly a useful one, but still must be
considered a band-aid.

In any case where you can have started to write flash and then have a
brownout, you will eventually get into trouble if you are
updating-in-place, or doing anything that doesn't follow a
write->verify->enable flow. Regardless of whether you have a test "if
(above_thresold) { ... }" (what happens if you brownout right after the
test ?) or if you put the chip in reset (what happens if you reset after
some of your writes, or part way through the erase ?) It is completely
possible to write flash access algorithms that work completely reliably
even in the face of brownouts & resets without any special hardware support.

There is plenty of anecdotal evidence that even without consciously writing
to flash during boot, it still manages to happen somehow; and I stand by my
prior assertions that this is not a known problem with the raw STM32
silicon.

I've been unexpectedly travelling for almost all of the last 3 weeks for
the day job, or I would have a proton hooked up to a remote controlled
power supply in a test fixture by now. Hopefully this weekend I'll be able
to get to that.

monkbroc, 2015-12-09:
I don't have a test case unfortunately. I sometimes get SOS red blinking reboot loop with a battery application or loss of EEPROM when the battery is low, but nothing reproducible. As soon as I connect a debugger the SOS loop stops and the photon boots ok.

I don't dispute that there might be a deeper power issue Andy. OTOH based on reading the EEPROM code it should not fail because of reboots during the write phase. I expect that the brown out reset config will make the power stable for erase and programming operations.

peterqu, 2015-12-09:
Hi @andyw-lala 

I am not sure if we can write an flash access algorithm that works even with brown-outs and resets.  It would be great if you can share some of your ideas.  A few things come to my mind that could help:

1) Use a external supervisor IC such as TI's TPS3702 to generate a Power Good (PG) signal.  Modify the flash memory algorithm to write or erase only when PG signal is active.  
2) Larger decoupling cap that gives STM32 enough charge to finish existing flash memory write/erase before going into BOR.
2) Apply ECC technique such as Reed-Solomon, BCH or LDPC to detect or correct for bit error.  But this would probably need to allocate extra space for redundancy (maybe 30%? depends on the error correcting capability).  

It doesn't sounds like there is an easy way to fix the flash algorithm.  One of my earlier jobs was to design flash memory at a large semi company.  I remember finding capacitor/battery inside SD card and SSD for the purpose of preventing flash corruption.

m-mcgowan, 2016-03-18:
Can we re-investigate this to try to uncover possible causes? We can investigate EEPROM emulation specifically, but also general flash erase/write operations, such as DFU or OTA update. In particular, the system attempts to update the bootloader after a system update - this could be a potential point of failure if the flash operations are not reliable.

m-mcgowan, 2016-06-07:
Nearly 3 months since my last comment - no news is good news they say :-), so let's assume that the EEPROM rewrite in 0.5.0 have rendered this issue resolved. Happy to re-open if it reappears.

closeted, 2016-06-24:
Reproduced on 0.5.1

technobly, 2017-04-11:
Re-opening since it appears there are still issues occurring with power loss and flash memory on Photons/P1/Electrons.  We are continuously investigating this and looking for reliable ways to reproduce these issues.  If you have any leads please update this community bounty thread: https://community.particle.io/t/bug-bounty-electron-not-booting-after-battery-discharges-completely/25043

Meanwhile we are implementing techniques in system firmware to mitigate flash memory corruption for critical settings like Wi-Fi credentials, Public Server Key and Device Keys.  Some of these mitigations are planned for 0.7.0-rc.1 firmware.