particle-iot/device-os #706: P1 dfu-util 0.8 does not read/write from External Flash

jakeypoo, 2015-10-20:
Trying to read/write to the external flash on the P1 in the same way that I used to on the Core on alt interface 1, I've run into some issues.  

`dfu-util -l` gives an interface I assume is the external SPI flash on the P1 
    Found DFU: [2b04:d008] ver=0200, devnum=10, cfg=1, intf=0, alt=2, name="@Serial Flash   /0x00000000/256*004Kg", serial="00000000010C"

However trying to write to this interface via `dfu-util -d 2b04:d008 -a 2 -s 0 -D test.bin` gives an error that I don't see with the same command to other alt interfaces : 
>Opening DFU capable USB device...
>ID 2b04:d008
>Run-time device DFU version 011a
>Claiming USB DFU Interface...
>Setting Alternate Setting #2 ...
>Determining device status: state = dfuIDLE, status = 0
>dfuIDLE, continuing
>DFU mode device DFU version 011a
>DfuSe interface name: "Serial Flash   "
>dfu-util: Only DfuSe file version 1.1a is supported
>dfu-util: (for raw binary download, use the --dfuse-address option)

Trying to read from this interface works, however the file that gets downloaded is not the external flash. 
In fact the file that gets downloaded seems to start at the same address as the last dfu-util upload operation performed : 
For example if I run
` dfu-util -d 2b04:d008 -a 0 -s 0x08020000:1024 -U test1.bin`
`dfu-util -d 2b04:d008 -a 2 -s 0:1024 -U test2.bin`
The two files are identical.

I've run some tests with sFLASH_... commands on the P1 and they seem to work. But I can't seem to get the dfu-util part to work.

m-mcgowan, 2015-10-20:
Thanks for this - it is a known issue. There is some missing code in the bootloader that isn't taking into account the "alternate" setting from DFU and instead tries to determine the memory from the address only. This failes since both DCT and external flash start with address 0. 

We have a PR for this which I'll cherry pick the appropriate commits and have this released for the next version.

jakeypoo, 2015-10-20:
Awesome! Thanks. 
Sorry this is unrelated, is this memory reserved for anything? or can we allocate the whole thing for whatever we need at the application layer?

I haven't been able to find much documentation on the external flash.

m-mcgowan, 2015-10-20:
Nothing is reserved in external flash, so feel free to use it for whatever you want!

m-mcgowan, 2016-01-15:
There are two issues I found here:

- writing to address 0 isn't supported by dfu-util at present (presumably an oversight.) They use set the address to a variable like uint32_t address, then test if an address has been given by comparing it for non-zero. Naturally a 0 address will fail this test.  (The line of code is here - http://sourceforge.net/p/dfu-util/dfu-util/ci/master/tree/src/dfuse.c#l629 - I'll put it on the backlog to submit a PR.)

- on the P1, the compiler optimizations were breaking DFU reads/writes in the bootloader. (In order to fit into the available space we use fairly agressive link time optimization - LTO. This was causing writes to fail - sometimes the flash wasn't erased, or data written.  I couldn't determine if the compiler was at fault or the code. The fix was to reduce the optimization level for the external flash routines.  

So for the time being, writing to address 0 isn't supported until it's fixed in . You can write to address 1 and up! ;-)

jakeypoo, 2016-01-15:
To clarify, dfu-util R/W to address 1 will access external flash address 1?

m-mcgowan, 2016-01-15:
Yes, that's correct. 

Just thinking about writing to address 0. As a workaround, we could modulo the address by the SPI flash size. The flash is 1MByte, so 1024*1024 bytes. Writing to that address would write to address 0 in the flash memory, and work around the address 0 bug in dfu-util.

commit 9ff86e29 (2016-01-15): reduces compiler optimizations which breaks DFU to Serial flash on the P1. fixes #706
closed 2016-01-15 by m-mcgowan