particle-iot/device-os #707: SOS when `Particle.variable` registered to returned `String` object

pomplesiegel, 2015-10-21:
Using the code below running on System FW 0.4.7, when you call 

`particle get <deviceName> systemFW`

we're seeing bad behavior in the form of 
-SOS's
-Value returned as either blank, gibberish, or representing the wrong `variable`

````cpp
#include "application.h"

SYSTEM_MODE(MANUAL);
SYSTEM_THREAD(ENABLED);

void setup()
{
  Serial.begin(9600);

  WiFi.on();
  WiFi.connect();
  Particle.connect();

  Serial.println("Example 'variable' program");

  Particle.variable("wifiSSID", WiFi.SSID());
  Particle.variable("systemFW", System.version());
}

void loop()
{
  Particle.process();
}
````

* `wifiSSID` works reliably and is totally fine, likely because the value returned by `WiFi.SSID()` is a `const char *`, a true c-string. 

* However, `System.version()`, which returns a populated `String` object, has undefined behavior causing misinformation or hard crashes. In our full-featured app, it returned either gibberish or the value of another registered `variable`.

dmiddlecamp, 2015-10-21:
Hi @pomplesiegel,

  The string in Particle.Variable should be a variable you've declared in your app, not the return value of a function call.  I suspect the return value is getting cleaned up and causing issues.  Try something like...

```
#include "application.h"

SYSTEM_MODE(MANUAL);
SYSTEM_THREAD(ENABLED);

String ssid;
String version;

void setup()
{
  Serial.begin(9600);

  WiFi.on();
  WiFi.connect();
  Particle.connect();

  Serial.println("Example 'variable' program");

 ssid = WiFi.SSID();
 version = System.version();
  Particle.variable("wifiSSID", ssid);
  Particle.variable("systemFW", version);
}

void loop()
{
  Particle.process();
}
```

Thanks,
David

m-mcgowan, 2015-10-22:
It's been on my mind to add a check for this at runtime. (I don't know how this can be done at compile time just yet.) 
We can know if the variable is allocated on the heap or the stack, so if it's on the stack we'll simply not register it and return false. (And add troubleshooting to the docs.)

pomplesiegel, 2015-10-22:
Totally, I see how if a `variable` is simply registered to -> a String object on the stack, it will likely be destroyed by the time the user asks for it, causing a hard crash upon a `get`.

I'm curious if you all have a better idea for how to implement OTA retrieval of these common System states using our current toolkit and API. 

Here's my concern: 
* many of these firmware-accessible System states like `System.version()` or `WiFi.SSID()` are things we often would like to know about OTA via a `get` for system debugging from afar. 
* Many of them change dynamically at runtime, so we need an up-to-date version of that state. For that reason, assigning `WiFi.SSID()` to our own String object and pointing our `variable` to that static String does not help us, as this is not an up-to-date copy.

`WiFi.SSID()` does not currently suffer from this issue because it returns the location of a `const char *`. While the value of the c-string pointed to by `WiFi.SSID()` may change at runtime, its memory location does not, allowing us to successfully register this as a `variable` whose value changes accurately upon a `get`.

Going forward, if many system states are returned as `String` objects on the stack, @dmiddlecamp's above solution of
````cpp
String myString = WiFi.dnsServerIP();
Particle.variable("dnsIP",myString); 
````
may not help us, as by the time we `get` the value of dnsIP, this static copy will likely be out of date.

In order to keep the value of `myString` fresh (so our `get`s are accurate and useful) we need to know when the value of `WiFi.dnsServerIP()` has changed. Without some sort of flag or callback for this event, we would have to constantly poll the value to look for a change.

Would it make sense instead to have `String` objects representing System states like `dnsServerIP` permanently allocated on the heap (even though their value may change) and have a method like `WiFi.dnsServerIP()` simply return a pointer to that unmoving String object?

andyw-lala, 2015-10-22:
< thinking out loud, while working on something else >

Could we do anything with pointers-to-pointers, to use old-school terms ?
Chasing those pointers each time ?

Nothing can help us where scope issues are involved; but when it is simply
a dyanamic object, perhaps adding a level of indirection could offer
something ?

How to make that user friendly, is an exercise left to the reader.

< />

m-mcgowan, 2015-10-22:
This is an interesting problem - it's been around since variables were introduced in firmware, that all variables should be global, although it is now made easier to run into with the convenience of `String` variables. 

For non-changing data, an immediate workaround is to copy the data to a global string instance:

```
String dnsIP;

void setup()
{
   dnsIP = WiFi.dnsServerIP().toString();
   Particle.variable("dnsIP", dnsIP);
}
```

Although this doesn't feel like a complete solution since even the DNS IP is subject to change under some circumstances.  That could be addressed by copying the data on each iteration of loop.

Note that I've made the conversion of `dnsServerIP` from an IPAddress to a String explicit. We could make the conversion implicit but I want to highlight the fact that we can't simply reference the original IP address and pass that as a pointer, since the variable is a derivation of that IPAddress value.

With the present API and design of how variables function, that's the solution available. Variables must have a lifetime as long as the variable registration, which typically means they must be in global scope. 

*Looking Forward*
I'm just thinking aloud here - so no promise this will be our chosen solution ;-) - but it feels like variables can be made available as function callbacks, so that the most recent value can be retrieved:

```
Particle.variable("dnsIP", []{ WiFi.dnsServerIP().toString(); });
```

The syntax is a little clumsy for anyone not familiar with lambdas - so a shortcut could be made available:

```
Particle.variable("dnsIP", from(WiFi.dnsServerIP().toString())); 
```

With an implicit conversion to String, this then becomes:

```
Particle.variable("dnsIP", from(WiFi.dnsServerIP())); 
```

This would ensure the latest version of the variable is fetched whenever variables are retrieved from a device. 

I know some folks have expressed concerns over the fact that variables are fetched via the Cloud API using a GET request, so how can callbacks have side affects and change a variable?  To my mind, the application coder can already change a global variable anytime they like, making variable values already subject to change. (The clue is in the name!)  So introducing callbacks doesn't change the present situation of non-idempotent GET requests. 

I know @dmiddlecamp has some ideas about changes to cloud variables - I look forward to hearing those sometime!

pomplesiegel, 2015-10-22:
@m-mcgowan, major :+1: for that lambda solution. This would also allow the firmware to generate the value of a `get`table variable only when it's needed, rather than constantly, fearing a `get` from above at any moment.

sergeuz, 2020-02-26:
`Particle.variable()` will support functions/lambdas in 1.6.0: https://github.com/particle-iot/device-os/pull/1998

closed 2020-02-26 by sergeuz