particle-iot/device-os #710: Garbage char in first serial message

zerous01, 2015-10-22:
Bug from https://github.com/spark/firmware/issues/682 is still here.

When running following code on photon

unsigned long lastTime = 0;
void setup()          
{
    Serial1.begin(9600);
}

void loop()  
{    
    unsigned long now = millis();     
    if (lastTime == 0 ||  ((now - lastTime) > 5000)){ 
        Serial1.println("!");
        lastTime = now; 
    }
}

I receive following output:

▒!
!
!

m-mcgowan, 2016-01-15:
Can you try adding pullup resistors to the TX/RX lines? (Thanks Andrey for the suggestion.)

m-mcgowan, 2016-01-28:
No activity.  Can be re-opened if the pull-ups don't help.

closed 2016-01-28 by m-mcgowan

zerous01, 2016-02-01:
My photon is mounted long inside   a train for several weeks :), and i have only one right now. Any possibilities you guys can test it with pull-ups?

Skickat från min Sony Xperia™-smartphone

---- Matthew McGowan skrev ----