particle-iot/device-os #713: Photon I2C on v0.4.7 hangs system with no pull up resistors

technobly, 2015-10-26:
This is a case we should never see because you should have pull up resistors when using I2C, however that doesn't mean a resistor might not fall off the board when in use (yes it can happen).

Figure out where the code is hanging with this routine, and add a fix.

Serial output never occurs and D7 remains low.
```c++
#include "application.h"

void setup() {
    Wire.begin();
    Serial.begin(9600);
    pinMode(D7, OUTPUT);
}

void loop() {
    Wire.beginTransmission(0x55);
    Wire.write(0x55);
    Wire.endTransmission();

    digitalWrite(D7, HIGH);
    Serial.println("YO");
    delay(1000);
}
```

m-mcgowan, 2015-10-26:
My guess is it's the timeout not working which is dependent on millis(), which might not be incremented with the recent addition of non-task switching. A fix would be to switch to micros() which does continue to increment with interrupts disabled.

andyw-lala, 2015-10-26:
IMO - it is important that this case (missing pull-ups) should be handled
gracefully. Obviously reliable data transfer will not take place, but
hanging is not acceptable.

technobly, 2015-10-26:
I'm not certain, but I seem to remember when heads down in I2C-land for 0.4.7 changes... I saw the BERR (Bus Error) flag set repeated when there were no pull up resistors.  So this could be leveraged possibly if the micros() change doesn't do everything we need it to already.

When you say "gracefully" @andyw-lala what's you idea of that vs non-gracefully?  I have an idea, but just wanted to see what you were thinking as well.

andyw-lala, 2015-10-26:
All access functions/methods return within a bounded time, with an
unequivocal error indication.

They can leave the bus in an unknown state, if that is the way it has to be
(forcing the user to re-init.)

dresslerc, 2015-10-26:
I just want to point out that I have pull up resistors on my MCP23017 and it locks up with 0.4.7.  Issue doesnt occur on 0.4.6.

I'll setup my test Photon tonight and see if I can get it to lock up.

jenesaisdiq, 2015-11-19:
Hey guys and @peterqu, easy reproduction on this with a Photon that isn't plugged into a circuit. The particular line that causes the hang appears to be Wire.endTransmission().

peterqu, 2015-11-23:
Hello @m-mcgowan, you are absolutely right.  Replacing millis() with micros() indeed resolved the I2C_HAL_End_Transmission hang issue. 

I think I will proceed to replace all millis() inside i2c_hal.c with micros().  Do you see any potential issue with replacing the all the millis() timer in i2c_hal?

m-mcgowan, 2015-11-27:
Should be fine! Get's my  :+1:

peterqu, 2015-12-01:
Hi, I have confirmed @technobly 's latest change for electron I2C has also fixed this issue.

m-mcgowan, 2015-12-02:
the commit is https://github.com/spark/firmware/commit/6a048f825d4838908cf2b97e4256895d797b8a7a

KevinRobertsonWright, 2015-12-03:
As a related issue, I've found that once I hang the system by running an I2C communication with no pull-ups (happens when the chip is detatched from the board as mentioned above), resetting or powering down the electron does not return it to a good state. After pushing new firmware, it seems to start behaving itself again.

technobly, 2015-12-03:
@KeighDub what system firmware are you running on your Electron?

KeighDub, 2015-12-03:
The system I’m running right now is 0.0.3. I’ve also seen it on the tip build from a week ago or so, but I’d have to look back to see which day’s build it was.

technobly, 2015-12-03:
@KeighDub thanks for that.  I think you might not have been testing with the fix that was merged into the develop branch 3 days ago.  If you are compiling locally could you try the latest develop branch and see if the problem still persists or not? https://github.com/spark/firmware/tree/develop  Thanks!

technobly, 2015-12-03:
FYI: I just tested this myself with the Electron & Photon with the original test app, with and without pull up resistors with the latest develop branch and everything works great :+1:  No lockups.

Going to close this, please let me know if you're not seeing the same results.

closed 2015-12-03 by technobly