particle-iot/device-os #714: allow late registering of functions / variables

dmiddlecamp, 2015-10-26:
Since the cloud caches the describe results, and also supports partial describe payloads, could we send unsolicited describe updates later on if a user registers new Functions / Variables?

m-mcgowan, 2015-10-26:
Sure. That's possible!  How about we send the describe in full after all functions have been registered (there is a way to detect this) so that we can also allow functions to be unregistered? (This has come up once in the forums recently.)

ScruffR, 2017-07-15:
To bump this up again and extend
It would also be good to have this work for `Particle.subscribe()` just the same.

And an individually targeted unregistration to free up slots for functions, variables or subscriptions could be desirable too.
Currently there is no way for functions and variables while for subscriptions you can only ditch them all at once.
A per instance unregistration might become handy.

brandoaire, 2017-07-17:
Being discussed in the community, here:
https://community.particle.io/t/photon-functions-and-variables-missing-after-no-changes/31520