particle-iot/device-os #715: photon : Comparison between IPAddress objects does not always work

jakeypoo, 2015-10-26:
Ran into a bug where comparison between a resolved IPAddress is not equal to an IPAddress created using the constructors. Take a look at the test code below : 

```
IPAddress test_a = IPAddress( 192 , 168 , 1 , 142 ) ; 
IPAddress test_b = IPAddress( 192 , 168 , 1 , 142 ) ;
IPAddress test_c = WiFi.resolve("192.168.1.142") ;

if( test_x == test_y ){         // Test 1 
    Serial.println( "x == y" );
}
if( test_y == test_z ){         // Test 2 
    Serial.println( "y == z" );
}
```
Test 1 above passes if `test_a` and `test_b` are declared globally, outside of any functions; otherwise both tests fail. 

My temporary workaround is just comparing individual octets
``` 
if( ( test_x[0] == test_y[0] ) &&
    ( test_x[1] == test_y[1] ) &&
    ( test_x[2] == test_y[2] ) &&
    ( test_x[3] == test_y[3] ) ){
        ....
}
```

bkobkobko, 2015-10-26:
You have some code shift from _a _b _c to _x _y _z but I think your idea is there.

You cannot run WiFi.resolve() to set a global variable outside of setup() or loop() since the network may or may not be running when you call it.  You can call WiFi.resolve() in functions when the network is up and running of course.

Does that explain what you are seeing or is there some other problem?

jakeypoo, 2015-10-26:
oops, thanks edited the first comment to fix the variable names.

As I mentioned before. The comparison passes if you compare each byte individually but not if you compare IPAddress objects. So the actual values are being set properly, the comparison just doesn't work.

m-mcgowan, 2015-10-26:
Thanks for the test case. I believe it's the version field possibly not being set by all control paths. I'll add this to our internal test suite so that we can be sure this doesn't appear again.

commit 390b30dd (2016-01-18): tests for retrieving BSSID.  Also fixes #715.
closed 2016-01-18 by m-mcgowan