particle-iot/device-os #719: Photon can get into a state where iOS app hangs at connect screen - unable to set up Photon wifi credentials until you reset

karlhenricksen, 2015-10-27:
### Description:
We have seen occasionally that during an attempt to set up new wifi credentials the app will hang indefinitely as it attempts to connect to the Photon wifi. Quitting the app and retrying does not fix the problem, and we are forced to reset the photon in order to successfully send wifi credentials. This has been hard to reproduce consistently, so while the the following steps are more specific (resending duplicate credentials), it produces the same behavior very consistently. No timeouts or Setup Failure screens are encountered even after waiting 5+ minutes for the connection to resolve (possibly a secondary issue).

### Steps to Reproduce:    
1. Flash Photon with the test user program code below
2. Set Photon into Listening Mode by holding right button 
3. Using wifi setup app, send correct router credentials to Photon
4. **Important** once Photon LED goes to breathing cyan, *immediately* close setup app
5. Without resetting Photon, repeat steps 2-4 a few times
6. If a failure/hung app occurs, close app and attempt to retry the normal setup process
* typically, I see it takes one or two times before the behavior is seen - setup app hangs indefinitely at the screen below, and subsequent attempts to setup have the same behavior, requiring a reset to fix.

[image]

Note that these steps are just the reliable way to reproduce, but I have also seen this during normal operation - ie. changing credentials and waiting for the app to fully complete the checklist. It just appears that when I exit the app early it is much more reliable, and I would think that shouldn’t cause the Photon to get into the state where you need to reset.

### Code to reproduce:

````cpp
#include "application.h"

SYSTEM_MODE(MANUAL);
SYSTEM_THREAD(ENABLED);

STARTUP(WiFi.selectAntenna(ANT_AUTO));

void setup()
{
  Serial.begin(9600);

  WiFi.on();
  WiFi.connect();
  Particle.connect();

  Serial.println("Example program");

  Particle.variable("wifiSSID", WiFi.SSID());
}

void loop()
{
  Particle.process(); 
}
````

System version 0.4.7

idokleinman, 2015-10-27:
Thanks for reporting this. I am not entirely sure this has something to do with the current user firmware running on the device (it is not running while in listening mode). We'll investigate this issue.
cc: @m-mcgowan

karlhenricksen, 2015-10-28:
Great, thanks. Good point on the user fw, just wanted to make sure you had a similar setup to remove our user pgm as a factor. And I assumed this to be a firmware issue rather than iOS since the Photon requires reset to recover (app quit/relaunch or even phone reboot doesn't help), but I see your point. Curious to how this happens though since we've been seeing this for a while and could not reproduce until now.

mebrunet, 2015-10-28:
I wonder if this isn't related to #692

m-mcgowan, 2016-03-14:
Any next steps we can take with this issue? Does it seem hardware related as in #692?

karlhenricksen, 2016-07-14:
@m-mcgowan Can we take a closer look at testing for wifi setup issues like this? We've had customers experience this particular issue and I've since dug into reproducing with tinker and confirmed that it is indeed reproducible. I also discovered #1061 through this testing with just tinker, just by repeatedly testing the use case of sending incorrect user credentials. Unfortunately both these issues appear to be possible on the first attempt too, so not a great experience, since in our case the user would have to unplug the product from the wall to fix (we wouldn't want them accessing or playing with the Photon). I'll send new steps in another comment, the old ones were a bit convoluted. Thanks!

karlhenricksen, 2016-07-14:
It may be superstitious, but the way I've been able to semi-consistently reproduce has to do with waiting some time with app open before sending incorrect credentials. However, it may be more likely that it is the Photon sitting in listening that is the problem.

**Note**: the quickest way to do this is to line up as many photons as you have available with Tinker, and go down the line, attempting to connect to each after they sit in listening mode for a while

**Steps to Reproduce (with Tinker):**
1. Reset Photon wifi credentials via setup button (hold 10+sec) for multiple Photons
2. Reset all Photons to start clean experiment
3. With the iOS app, go to the wifi setup screen (above in main description, "1. Tap Home > Settings > Wifi etc" 
4. Wait for 10-12 mins on that screen, with Photons in listening mode
5. Attempt to connect to the Photon soft AP by tapping on it in iPhone Wifi settings
6. If app connects and proceeds to the Wifi Network list screen, kill app and retry with the next available Photon or wait more time to retry

**Version:**
System v0.5.2, user = tinker

technobly, 2017-02-07:
@karlhenricksen there were a number of bugs fixed with SoftAP after 0.5.2.  Would you mind retesting this with 0.6.1-rc.2 firmware?

technobly, 2017-02-22:
@karlhenricksen any help you can provide would be appreciated.  Thanks!

karlhenricksen, 2017-02-22:
Hi @technobly, sorry for the delay. I'm no longer working with Grove so unless something changes with that in the near future I probably won't be able to help on this one. Good luck, wish I could be of more use!

technobly, 2017-02-22:
Thanks for the heads up @karlhenricksen

closed 2020-02-28 by avtolstoy