particle-iot/device-os #724: Photon delays entering listening mode after router settings change (ex. new pw) for up to a minute

karlhenricksen, 2015-10-30:
### Description:
Photon delays entering listening mode after router settings change (eg. new pw) for up to a minute if router settings have been changed previously without a reset. It's possible this will only occur if the user changes their router password and soon after (under 2min) proceed to also update their Photon wifi credentials to match - not out of the question. Again, this may have been around in earlier versions and we have seen this before, but have just now found reproducible steps. It's not clear which of the elements are essential, but the steps below can reliably reproduce now. 

### Steps to Reproduce:    
1. Ensure Photon is connected to wifi and cloud with correct router credentials (breathing cyan)
2. Open your router settings config web/external app
3. Under Wireless Settings, change the router password and hit apply/submit
4. Once Photon starts blinking green due to incorrect creds, wait for approx 15sec (up to a minuter or 2max) before going to step 5
5. Press right button on Photon for 3-5 sec and release
6. If Photon goes to blinking blue Listening mode, send the new wifi credentials with iOS app to repeat step 1 and get it back online
7. Repeat steps 2-5 for another new password --> typically the issue will present itself the 2nd time
8. Wait up to 1min30sec for blinking green LED to change to blinking blue, indicating Listening mode

Note: this issue is somewhat exacerbated by #690 (behavior 1) since there is no LED feedback that the Photon will eventually enter listening mode, which may lead user to hold longer than 5sec, resulting a failure to enter listening mode at the end of the delay caused by this issue.

### Video
Video below shows the described behavior, starting from online to pw change (blinking green) to holding button for 4 sec (25sec point), followed by the Photon finally entering Listening mode at 1min31sec point.

http://www.youtube.com/watch?v=ndjuEP8RRds

### Dumb Test Code Used to Reproduce:

````cpp
#include "application.h"

SYSTEM_MODE(MANUAL);
SYSTEM_THREAD(ENABLED);

STARTUP(WiFi.selectAntenna(ANT_AUTO));

void setup()
{
  Serial.begin(9600);

  WiFi.on();
  WiFi.connect();
  Particle.connect();

  Serial.println("Example program");

  Particle.variable("wifiSSID", WiFi.SSID());
}

void loop()
{
  Particle.process(); 
}
````

System version 0.4.7

karlhenricksen, 2015-10-30:
I've also seen where holding the right button for 3-4 sec after changing wifi settings to put into listening mode turns the LED off for up to a minute before LED finally blinks blue.

m-mcgowan, 2016-01-22:
I tested against https://github.com/spark/firmware/commit/bc7d59b65bb8e22714b8d0186c5023747b91a10f and unfortunately couldn't reproduce the issue. If you are able to build locally, please try with that commit. (Alternatively, 0.4.9 will be released later today.)

karlhenricksen, 2016-04-21:
@m-mcgowan sorry for delay reproducing, but just checking against this with very simple user pgms and or product's user pgm and I can't reproduce this, against 0.5.0rc2. Since much has changed, I'm not surprised and will keep an eye out, but we haven't seen this issue for a while in normal operation - Probably safe to close if you don't see anything suspicious that might have caused this.

m-mcgowan, 2016-06-07:
Thanks, I'll close this out - we can re-open if new evidence comes to light.

closed 2016-06-07 by m-mcgowan