particle-iot/device-os #731: 0.47 -Semi-automatic wiced Timeout causes Photon/P1 to hang

jakeypoo, 2015-11-06:
I've been seeing some of my devices hang, and I think this is what's causing the issue. It's been happening very rarely and it seems to correlate with poor wifi reception or an AP going down for a few minutes. If it's the same bug, I believe it has affected some units running 0.46 as well. 

If you run the code below and reset/remove your wifi router so that the photon loses wifi connection, if the result of socket_send is ever -2 (wiced timeout) the device will hang. 
It won't enter listening mode if you hold down the setup button, or re-enter the user `loop()`

The same code seems to recover just fine in `AUTOMATIC` mode. I have not tried `MANUAL`.  

It's a bit of an esoteric bug, but might come up for those of us running in `SEMI-AUTOMATIC`

The `wdtimer` is not necessary, it's just a little extra there to reset the device after it hangs. 

~~~
#include "application.h"

SYSTEM_MODE(SEMI_AUTOMATIC); 

IPAddress test_target = IPAddress( 192 , 168 , 1, 142 );
UDP test_sock ;

void tRunner(void );
Timer       wdtimer( 2000UL , tRunner );

volatile int32_t    watchdog = 100;

char        msg1[] = "TEST 1\n\r" ;

void tRunner( void ) {
    
    watchdog-- ; 
    
    if( watchdog < 1 ){
        Serial.print( "WATCHDOG EXPIRED -- REBOOTING ..." );
        delay(50); 
        System.reset() ;
    }
    return ;
}

void setup() {
    Serial.begin(9600);
    WiFi.connect() ;
    wdtimer.start() ;
}

void loop() {
    static int32_t udp_open = -1 ;
    int32_t udp_result ;

    if( udp_open < 1 ){
        udp_open = test_sock.begin( 50000 );
    }
    else{
        Serial.print( "udp > " ) ;                                                                                                                                                                                                                                             
        test_sock.beginPacket( test_target , 4000 );       
        test_sock.write( (const uint8_t *)msg1 , strlen(msg1));
        udp_result = test_sock.endPacket() ;
        Serial.println( udp_result );
        if( udp_result == -2 )
            Serial.println( "Wiced Timeout" ) ;
    }
    watchdog = 100; 
    delay(50); 
}
~~~

m-mcgowan, 2017-07-07:
Much time has passed :-)  The winds have shaped the landscapes and the world looks very different now. 

I wonder if this is still an issue for you, and if it is, would you please provide further details that would help us investigate. Thanks!

jakeypoo, 2017-07-07:
Sorry, but I can't say that I remember. It's been a while since I worked on this firmware. 
You could try the above example code and see what happens. 

Best of luck.

closed 2017-07-07 by jakeypoo