particle-iot/device-os #737: Compilation Environnement Installation Windows

gbrault, 2015-11-15:
The list of dependencies is bigger than the one documented: head is needed for example etc...
I succeeded to get it work with the installation of MINGW. (the MSYS bin path needs to be added)
Only thing that didn't worked was CRC32, but I got one which seems to work from http://esrg.sourceforge.net/utils_win_up/md5sum/.
However I need confirmation this is working? it produce a checksum in hex and in () a number in decimal.
The dependencies documentation should be revised to avoid unnecessary waste of time. You should also recommend which is the "native" development platform (my guess LINUX?).
Thanks

peterqu, 2015-12-03:
Hi,
I think we have more developers using Mac as the native development platform. So the tool chain has less issues in Mac/Linux enviornment.  That being said, I have a working developing environment on MINGW in Win10.

gbrault, 2015-12-03:
Hello,
The above mentioned CRC32  program is working. I was able to flash a photon.

closed 2015-12-03 by gbrault