particle-iot/device-os #738: Soft delays in setup() can  prevent variable registration

bprobbins, 2015-11-15:
//two or more soft delays 
//(either of which is at least 1000ms in length) 
//in setup prevents registration of variable

#include "application.h"

char astr[64];

uint32_t lastTime;

void setup() {
    
    lastTime = millis();
    while (millis()-lastTime < 500) {Particle.process();} //500 ok, 1000 not
    
    lastTime = millis();
    while (millis()-lastTime < 1000) {Particle.process();}//500 ok, 1000 not

    Particle.variable("vartest2", astr, STRING);
}

void loop() {

}

m-mcgowan, 2015-11-15:
I think this is because the delays cause the variables to be added after the describe message has been sent so the cloud is not aware of these new variables. @dmiddlecamp  - any thoughts on how we can resolve this?  Some initial ideas:

- have the cloud send a describe request any time an unknown variable/function is requested.  
- have the device send a new describe message any time variables/functions are changed.

That's my initial thoughts, look forward to hearing your thoughts too.

dmiddlecamp, 2015-11-16:
So the cloud will send the describe message at least twice if the first response comes back empty.  So it's expecting either there was a delay, and nothing was registered, or everything was registered right away.  I think the easiest solution would be for the firmware to send an updated describe payload for the functions / variables if a new fn/var is registered after a describe response is sent to the cloud, etc.

Caching the response is a huge optimization, so I'd love to not lose that.  Thanks!

m-mcgowan, 2015-12-14:
So the fix here is that we send the describe message as functions/vars change since the last describe message was sent to the cloud (and the cloud is connected, i.e. post handshake.)

One detail I'd like to avoid is that when several variables/functions are registered, the describe message isn't sent after each registered item, but rather after they are all registered. This can be achieved by sending the describe message from the system background loop.

m-mcgowan, 2016-01-11:
This is a bigger change than I originally envisaged and requires cloud-side changes too. So we have to postpone this a little.

m-mcgowan, 2016-04-12:
The system keeps track of a checksum for the funcs/vars last sent to the cloud. This can be used to determine if the local state has changed since it was sent to the cloud.  If the local state has changed, then a new describe message is sent to the cloud containing the latest functions and variables. This is handled as part of the usual cloud communications protocol background processing.

The cloud will continue to send a describe message shortly after the hello message. The system will wait for that describe request from the cloud before sending any describe requests itself. This will reduce data use and will mean that for the common case of no delays in setup(), that the system behavior is unchanged.

m-mcgowan, 2016-06-07:
The checksums for the data last sent to the cloud is part of 0.6.0, but attempting to send a describe message from the system (unsolicited) resulted in the cloud connection becoming disrupted. Will schedule for the next release. cc: @dmiddlecamp, @brycekahle for a heads up on allowing the device to send it's own describe message without first being prompted from the cloud.

tiborrr, 2020-04-30:
Is this issue still under review? I'm experiencing the same issues with registering functions.

sergeuz, 2020-04-30:
@tiborrr It's fixed in `develop`, see PR https://github.com/particle-iot/device-os/pull/2024:

> In the AUTOMATIC mode, the setup() function is not executed until the connection with cloud is fully established. In this mode, the application's Describe message will be sent immediately after the application returns from setup(). Depending on the threading mode, this may happen either during the handshake with the cloud or immediately after it.

I can't provide the exact timeline for when the fix is actually going to be released though.