particle-iot/device-os #739: Proposed: à la carte debugging

technobly, 2015-11-16:
Currently you pick a debugging level and anything including that level and higher is output, like this:

```
SerialDebugOutput debugOutput(9600, ALL_LEVEL); // all levels

// or 

SerialDebugOutput debugOutput(9600, ERROR_LEVEL); // error and panic levels
```

Right now INFO is higher than DEBUG, so you cannot turn off the high amount of information output while also getting DEBUG messages.  I have moved this in local code to a lower level, but I could see a case where this alone might not be enough for a user.  Being able to pick and choose which types of debugging messages we want to see would be the best, and then we don't have to debate the importance and order of these messages.

This can be achieved fairly easily by changing the debugging levels to bit flags instead.  Then flags can be OR'd together to build a custom debugging session.

### Current Enumeration:
```
typedef enum LoggerOutputLevel {
    DEFAULT_LEVEL   = 0,        // used to select the default logging level
    ALL_LEVEL       = 1,
    TRACE_LEVEL     = 1,
    LOG_LEVEL       = 10,
    DEBUG_LEVEL     = 20,
    INFO_LEVEL      = 30,
    WARN_LEVEL      = 40,
    ERROR_LEVEL     = 50,
    PANIC_LEVEL     = 60,
    NO_LOG_LEVEL    = 70,        // set to not log any messages
} LoggerOutputLevel;
```

### Proposed Enumeration:
```
typedef enum LoggerOutputLevel {
    DEFAULT_LEVEL   = 0xFFFFFFFF,        // used to select the default logging level
    ALL_LEVEL       = 0x0000007F,
    TRACE_LEVEL     = 0x00000040,
    LOG_LEVEL       = 0x00000020,
    DEBUG_LEVEL     = 0x00000010,
    INFO_LEVEL      = 0x00000008,
    WARN_LEVEL      = 0x00000004,
    ERROR_LEVEL     = 0x00000002,
    PANIC_LEVEL     = 0x00000001,
    NO_LOG_LEVEL    = 0x00000000,        // set to not log any messages
} LoggerOutputLevel;
```

Then you can do things like this:
```
SerialDebugOutput debugOutput(9600, DEBUG_LEVEL | PANIC_LEVEL);

// or

SerialDebugOutput debugOutput(9600, INFO_LEVEL | WARN_LEVEL | PANIC_LEVEL);
```

m-mcgowan, 2016-01-22:
I like this idea. I've been thinking along similar lines. As well as using a bitmask to filter log level, also filter by subsystem, so that we can focus on logs from only one particular area.  E.g.

enum LogCategory
{
    
     /* Network state and management - connected/disconnected, DHCP etc.. */
     LOG_NETWORK_STATE = 1<<0,
     /* Log socket operations */
     LOG_NETWORK_SOCKETS = 1<<1,
    /*  Log socket data */
     LOG_NETWORK_DATA = 1<<2,
   /* Changes to cloud state, connected, disconnected, handshake, timeout etc... */
     LOG_CLOUD_STATE = 1<<3,
   /* Log cloud protocol */
     LOG_CLOUD_PROTOCOL = 1<<4,
    /* Log system events as they happen */ 
    LOG_SYSTEM_EVENTS = 1<<5,
     /* details arising from SYSTEM_THREAD() */
     LOG_SYSTEM_THREAD = 1<<6,
     
};

Adding a category to each log statement would be a lot of work. We can simplify this initially by defining the category based on the module (hal, system, communications etc..) This will give a rough categorization. 

Source files can set the default category for logs in that file by defining a symbol at the top of the file, e.g. `#define LOG_CATEGORY  LOG_CLOUD_PROTOCOL`

Finally, individual logging statements can specify the category:

```
INFO(LOG_CLOUD_STATE, "Connected to cloud hyperplane. Distributed AI online with %d Zettabytes of accumulated experience.", 12);
```

With each logging statement given a category, logging output can then be filtered by both level and category:

```
SerialDebugOutput debugOut(9600, ALL_LEVEL + LOG_CLOUD_STATE);
```

sergeuz, 2016-01-29:
Don't the approach with "flagged" log levels look a bit counter-intuitive? I never seen that having something like `Logger myLogger(LOG_LEVEL_INFO)` would result in only info messages being logged, and not errors - in any logging library I worked with.

Typically, if you need more fine-grained control over what is logged and what is not, you have to deal with logging contexts. Taking Mat's approach with log categories and C++ initializer lists, we could do something like this:

```
SerialDebugOutput debugOutput(9600, WARN_LEVEL, { // Default logging level
    {LOG_NETWORK_STATE, TRACE_LEVEL}, // Context-specific logging level (optional)
    {LOG_NETWORK_SOCKETS, TRACE_LEVEL}
});
```

Personally, I would go further and make logging context a string. Sticking to some naming scheme, we could simply write:

```
SerialDebugOutput debugOutput(9600, WARN_LEVEL, { // Default logging level
    {"hal.inet", TRACE_LEVEL}, // Context-specific logging
    {"system.cloud.*", TRACE_LEVEL} // Prefix-based matching (wildcard is allowed only at the end of the string for simplicity)
});
```

It's also quite handy to use string-based contexts during development, so you won't need to extend `enum LogCategory` every time you work on some internal subsystem that doesn't deserve its own category ID (like Brett currently have to do for log levels).

m-mcgowan, 2016-02-05:
I like that API. I think the filters should be prefix by default, so no need for `.*` at the end. 

If we can be sure only one instance of the context string is stored in flash then this can work. 

I'd also like to look at what standard logging formats are available and corresponding logging viewers so that users don't need to filter logs at runtime, but instead log everything and then filter at presentation time after loading the log into a viewer (or filtering via grep.)

sergeuz, 2016-02-12:
Guys, please check this commit: https://github.com/sergeuz/firmware/commit/29b45a720218ddb99eed685027e9809b32ba26b1. Key changes are:

- String-based log categories with per-category level filtering. Current implementation splits category strings by `.` character and uses binary search to determine category-specific logging levels. The approach is a a trade-off between memory usage and performance comparing to possible implementations based on binary tree, hash or linear prefix-based lookup.

- Message formatting is moved to application code. It's possible for a user to implement custom logger and/or formatter.

- It's possible to instantiate multiple loggers with independent filtering settings.

The implementation is work in progress, and, if general design is ok, I'd like to discuss some of the current limitations:

- We seem to use both message-based and stream-based logging thorough the code - in latter case debugging output is dumped as is without any intermediate formatting (which is good when you can't afford wasting memory for formatted strings). While both can be easily mixed when something like serial console is used as a backend, it's not clear whether the same is applicable for e.g. cloud-based logging. Current implementation handles message- and stream-based logging separately.

- It seems to be quite expensive to perform category lookup when stream-based logging is performed in per-character manner, e.g.: https://github.com/spark/firmware/blob/develop/communication/src/dtls_message_channel.cpp#L468. Current implementation doesn't perform category filtering for stream-based logging. As a possible solution that doesn't affect existent code entirely, we could cache last used category string and its level for each logger (possibly using TLS if threading is enabled), but I'd like to know if I'm overthinking here.

- Currently log categories can be defined by `LOG_CATEGORY(name)` macro which simply unwraps to `static const char* const _LOG_CATEGORY = name` (`static` is used to comply with both C and C++ that use different linkage for constants). This definition can be safely included by all files of particular module (either manually or via gcc's `-include` flag) and all string constants duplicated in different translation units will be merged during linking phase. The same `LOG_CATEGORY()` macro can be used to override logging category for particular scope, such as function or class. I think it also would be nice to have possibility to define logging category for particular source file, but I didn't manage to find suitable approach yet. Any ideas?

closed 2016-08-09 by technobly