particle-iot/device-os #743: Photon remains in listening mode after WiFi.listen(false)

airbornemint, 2015-11-23:
My code (which I haven't tried boiling down to a minimal example yet) is running on a Photon with SYSTEM_THREAD(ENABLED); and firmware version 0.4.7 installed.

I call WiFi.listen(true) to put the Photon into listening mode, and it does so correctly (on-board LED switches to blue flash, and Photon WiFi AP shows up on my nearby iPhone). 

I later call WiFi.listen(false) to take the Photon out of listening mode, but nothing changes.

Is there a known problem with WiFi.listen(false) not working?

BracWebb, 2015-11-23:
I have had that same issue.

I also had an issue where `waitUntil(WiFi.hasCredentials);` would move on immediately tho `WiFi.hasCredentials` was false. I was trying to do the following:
```
WiFi.listen(TRUE);
waitUntil(WiFi.hasCredentials);
```
before moving on in multithread. I actually still wanted to block the application.

I hacked around all that with:
```
WiFi.listen(TRUE);
waitFor(WiFi.ready, firstWiFiCredentialsTimeout);
System.reset();
```
(added a timeout, but main point being that I had to wait for `WiFi.ready` which can only occur once the `WiFi.hasCredentials`)

pomplesiegel, 2015-11-24:
I observed the same issue a few days ago. @airbornemint, Thanks for filing this.

m-mcgowan, 2015-12-23:
@sergeuz is looking into this! thanks!

closed 2016-01-19 by m-mcgowan