particle-iot/device-os #746: Software timers sometimes fail to start.

jakeypoo, 2015-11-23:
I have encountered cases where the system timers fail to start, this seems to happen after a system fault that causes a soft reset. 

My code currently uses 5 software timers and occasionally I see number 4 or 5 not run at all after calling start. According to the docs, the limit on soft timers is 10. 

Invoking a reset using the `System.reset()` function seems to recover fine. So I believe what's happening is something causes the program counter to reset, but the memory is not reset and thus Timer instances are still running from before? The timers perhaps reach the 10 timer limit and won't start any more.  

This condition can be checked by making the following modifications to the `spark_wiring_timer.h` file.
Add two public variables in the Timer class and check them on boot. I check the return values for the Timers I start to ensure they work.
Both functions should return 0 on success, occasionally I see a 1 on the `start_rv` return value.  
```cpp

public:
...
   Timer(unsigned period, timer_callback_fn callback_) : handle(nullptr), callback(callback_) {
        start_rv = -1 ;
        create_rv = os_timer_create(&handle, period, invoke_timer, this, nullptr);
    }
    int  create_rv ;
    int  start_rv ;
...
    void start(bool fromISR=false)
    {
        stop(fromISR);
        if (handle)                                                                                                                   
            start_rv = os_timer_change(handle, OS_TIMER_CHANGE_START, 0, 0, fromISR, nullptr);
    }
```
Not sure if it's a symptom of a bigger problem somewhere, but maybe part of the startup process should be to free any running timers?

m-mcgowan, 2015-12-14:
Are you sure you are releasing the previously allocated timers?

jakeypoo, 2015-12-14:
I initialize the timers only once in my application and never release them (they should run forever). When I see the `Timer.start()` fail it's something that happens about once in 10 restarts. 
My guess is some variable in the free-rtos/timers is not initialized properly and can have garbage in it on reset. I'll see if I can create a test-stub that can reproduce the issue.

m-mcgowan, 2016-01-22:
Just checking in to see if any more info has come to light about this?

jakeypoo, 2016-01-22:
Not much time to look into this issue, unfortunately. 
It is still occurring, but I just restart the micro if I see a non-zero return value from `os_timer_create` or `os_timer_change`

m-mcgowan, 2016-01-22:
The time periods - is there any chance they could be 0? A  timer with a period of 0 will fail to create.

jakeypoo, 2016-01-22:
Time periods are all non-zero and hard-coded. 
It's a bug that only occurs sometimes, but always the 4-5th timer that fails to start in my application. 
And it always seems to be the `os_timer_change` return value that fails.

m-mcgowan, 2016-01-22:
Thanks that was useful. I think I see the problem - the timer creation API includes a duration for how long to wait for the timer to be created.  Normally no delay is needed if the timer service is idle, but if the timer is being serviced then the calling thread may need to block. 

Currently the delay is hard-coded to 0. I think opening this up to the Timer api, with a default of block indefinitely will provide more reliable timer creation, and allow advanced coders to tune the blocking time to meet the dynamics of their app.

m-mcgowan, 2016-01-23:
A fix pushed in PR #834. Please try this if you have chance.

closed 2016-01-28 by m-mcgowan

jakeypoo, 2016-04-19:
Looks like this has indeed fixed the issue. Thanks and sorry for the long delay.