particle-iot/device-os #748: WLAN interface should not reset after failed cloud connection attempts in MANUAL mode.

dfukumori, 2015-12-01:
The current behavior is to reset the WLAN interface after MAX_FAILED_CONNECTS (currently 2) failed attempts to connect to the cloud. This alters the state of WiFi connectedness, which doesn't seem like proper behavior for MANUAL mode.

m-mcgowan, 2016-01-17:
Hi @dfukumori!

Thanks for reporting this issue. Can you say why you think it shouldn't drop the wifi connection in MANUAL mode?   Once Particle.connect() has been called, the system will try connecting to the cloud regardless of mode. If the cloud isn't available, which might be due to problems with WiFi then the system reinitializes the WiFi connection in an attempt to get things moving. 

Ideally the system shouldn't touch the wifi connection itself, but reality means this is sometimes necessary to preserve the connection. 

I see both sides of the story here and welcome discussion so we can work out what the right course of action is.

m-mcgowan, 2016-03-18:
Pulling the network connection on failed cloud connect is disruptive, especially if there are other connections working fine, and may be unnecessary if the reason for the failed connect is invalid keys, intermittent connectivity etc.

I propose we make this a system flag `SYSTEM_FLAG_BOUNCE_NETWORK_ON_CLOUD_UNRECHABLE`, with the default value being 1 (to preserve the current behavior.) In a later release, we then change the default to be the opposite - so the system doesn't bounce the network connection, which IMHO is less surprising. 

The default may also be platform-specific, since I believe this was mainly a workaround for connectivity problems on the Core.

closed 2016-08-09 by technobly