particle-iot/device-os #751: System.enterSafeMode() does not work

kennethlimcp, 2015-12-02:
Tested on 2 Photon with 0.4.7 firmware:

```
void setup(){

}

void loop(){
      System.enterSafeMode();
}
```

monkbroc, 2015-12-15:
Make sure your Photons have at least bootloader version 5.

This program fails to enter safe mode for bootloader 4 but works fine for bootloader 5.
```
#include "application.h"

void setup() {
  Serial.begin(9600);
  delay(200);
  Serial.println(HAL_Bootloader_Get_Flag(BOOTLOADER_FLAG_VERSION));
  delay(200);
}

void loop() {
  System.enterSafeMode();
}
```

m-mcgowan, 2015-12-15:
That's a great point. The system firmware that this was included should update the bootloader, iirc. @kennethlimcp  what version of system firmware is this running against?

monkbroc, 2015-12-15:
[bootloader_platform_6.c](https://github.com/spark/firmware/blob/954e7ca0464fd976a868f7f99abeeb605c348289/hal/src/stm32f2xx/bootloader_platform_6.c) hasn't been updated in 3 months. That's the mechanism for automatic bootloader updates, right?

kennethlimcp, 2015-12-15:
```
{"p":6,"m":[{"s":16384,"l":"m","vc":30,"vv":30,"f":"b","n":"0","v":4,"d":[]},{"s":262144,"l":"m","vc":30,"vv":30,"f":"s","n":"1","v":8,"d":[]},{"s":262144,"l":"m","vc":30,"vv":30,"f":"s","n":"2","v":8,"d":[{"f":"s","n":"1","v":8,"_":""}]},{"s":131072,"l":"m","vc":30,"vv":30,"u":"C99D612B3E3746B6FA32AD031F24D114E124055FF19DA62C85B1E339683D0105","f":"u","n":"1","v":3,"d":[{"f":"s","n":"2","v":8,"_":""}]},{"s":131072,"l":"f","vc":30,"vv":0,"d":[]}]}
```

commit f53b09ce (2016-01-18): update latest bootloader. fixes #751.
closed 2016-01-18 by m-mcgowan