particle-iot/device-os #753: Application DCT location

XuGuohui, 2015-12-03:
Hi there,

In the file dct_hal.h , it implies the application dct is placed at offset of `sizeof(platform_dct_data_t)`+ 1KBytes. But the implementation of `dct_read_app_data()` and `dct_write_app_data()` imply the application dct is placed just at offset of `sizeof(platform_dct_data_t)`.

So what scheme would you prefer in the end?

Cheer!

m-mcgowan, 2015-12-03:
In WICED, the DCT comprises 2 parts - the system data and the application data. The system data comes first, and followed immediately by the application data. 

`dct_read_app_data()`/`dct_write_app_data()` read and write to offsets in the application data region (in a platform-specific way.) On WICED, this reads and writes to the data after the system data. The indices passed to these functions are offsets in to the application data, so the first byte of application data is at index 0, even though this is at a much larger index in the actual WICED DCT region, since it comes after the system data. 

The 1KByte offset in the dct_hal.h struct is not used - it was an early attempt to inject some room between the system data and the application data so that we could tolerate a change in system data size. 

I hope that clears up the confusion!

XuGuohui, 2015-12-03:
Get it, thanks!

closed 2015-12-03 by XuGuohui