particle-iot/device-os #758: Implement Thread.elevatePriority

monkbroc, 2015-12-07:
https://community.particle.io/t/system-thread-can-it-be-disabled-at-runtime/18075

For critical but long-ish bits of application code it would be useful to temporarily elevate the priority of the current thread.

Proposed API: `Thread.elevatePriority` should elevate the priority for the current block and restore it when falling out of scope.

m-mcgowan, 2016-01-12:
We have the CriticalSection RAII class, which stops thread scheduling but does't prevent interrupts. That would seem to cater for this need without having to juggle priorities?

I'm not entirely happy calling this `CriticalSection` since while it does exclude other threads from entering that section of code, it does it in a very specific way. I think I'd prefer to rename it `SingleThreadedSection` to more accurately describe the mechanics of what is happening. 

I can envisage a family of `xxxSection` classes for different levels of exclusivity:

- `SingleThreadedSection` - disables thread scheduling but not interrupts for the current code block
- `AtomicSection` - disables thread scheduling and interrupts for the current code block
- `PrioritySection` - raises the priority of the current thread above background threads (but allowing critical services to still run.)
- `CriticalSection` will then be repurposed to be a traditional critical section involving mutexes which doesn't alter any scheduling or interrupts. 

I'd also like to make sure that whatever we use can be easily migrated to the Wiring++ API (or that we use Wiring++ spec from the start.)

monkbroc, 2016-01-12:
Well reasoned. This is only one part of the discussion on the threading API.

m-mcgowan, 2016-01-21:
We'll come back and look at thread priority as a separate feature in conjunction with reviewing Thread class API.

closed 2016-01-21 by m-mcgowan