particle-iot/device-os #761: Particle.publish() hangs on Photon when called in listening mode with SYSTEM_THREAD(ENABLED)

dfukumori, 2015-12-09:
I'm using the following to test:

    #include "application.h"

    SYSTEM_MODE(MANUAL);
    SYSTEM_THREAD(ENABLED);

    #define PUBLISH_PERIOD_MS 2000

    int last_publish_attempt_ms;

    void setup() {
      Serial.begin(9600);
      Particle.connect(); // Should be run without credentials present.
      last_publish_attempt_ms = 0;
      delay(2000);
    }

     void loop() {
       if (millis() - last_publish_attempt_ms >= PUBLISH_PERIOD_MS) {
         last_publish_attempt_ms = millis();
         Serial.println("pre-publish");
         Particle.publish("test");
         Serial.println("post-publish");
       }
     }

When monitoring the serial output over USB, "pre-publish" is printed, but "post-publish" is never printed. Furthermore, the blinking blue light of listening mode becomes very slow after ~10 seconds.

ScruffR, 2015-12-10:
`Particle.publish()` should check for `Particle.connected()` (or any privat flags for that reason) and just return `false` if there is no connection.

m-mcgowan, 2015-12-14:
Sure. for now I agree that Publish should fail if in listening mode or the cloud is not connected. In future, we may use different strategies, such as queuing the publish until the cloud is connected.

closed 2016-01-19 by m-mcgowan