particle-iot/device-os #764: describe message content is configurable from the server

m-mcgowan, 2015-12-14:
The TCP comms protocol has a 'd' CoAP message that is initiated from the cloud to request the device sends the system and application descriptors. The device can send these combined as a single message, or spread over several messages. 

With CoAP, sending these over several messages is not workable since this doesn't fit within the single request/single response CoAP messaging model. 

For UDP, we will need separate requests for system and application descriptors. Having these separate also helps conserve bandwidth, since the application descriptor is required each time a function/variable is added. while the system descriptor is needed only once. 

I suggest one of these representations:
- make 'd' return only the application descriptor. A new message 'D' returns the system descriptor. 
- add a query parameter to the 'd' request that encodes the type of descriptor data, with the default being nothing. (So application code doesn't accidentally overlook setting this parameter.) 

I would lean towards the 2nd one since it is more open-ended allowing additional types of descriptor data as they become available. 

cc: @brycekahle

brycekahle, 2015-12-14:
Query param sounds good. `s` for system, `a` for application?

brycekahle, 2015-12-14:
Do we have example contents for both responses?

m-mcgowan, 2015-12-16:
The 'd' message can take an optional additional location path that specifies the describe flags. The path value is a single byte combining the fags for the type of describe output required:

- system describe - value 1
- application describe - value 2

These can be combined to give combinations of output, e.g. value 2+1 = 3 for both application and system describe in the same response. 

If the describe flags option is not provided, the default is all descriptors. 

All types of descriptor results are valid json.

closed 2015-12-16 by m-mcgowan