particle-iot/device-os #769: Core: Building with Web IDE against "Latest (0.4.7)" fails - against "0.4.7" works???

ScruffR, 2015-12-14:
I've just had a member reporting problems with `Particle.xxxx` but `Spark.xxxx` worked.
After confirming he was targeting latest firmware, I tested myself and found that this error pops up when building against `Latest (0.4.7)` but does build fine against `0.4.7` (not Latest).

https://community.particle.io/t/particle-was-not-declared-in-this-scope/18285/4

monkbroc, 2015-12-15:
Looks like Latest is not pointing to 0.4.7. Here's another strange compiler error because of this:
http://community.particle.io/t/random-compiler-errors-on-system-thread-enabled/18301

monkbroc, 2015-12-16:
This is a build server issue. Dave knows about it now.

ScruffR, 2015-12-17:
**Please update the status of this issue!**

Currently Photons should work as usual again.

On Particle Dev and CLI version selection is not available, so the only alleged way around this is to try to build over and over till you hit a working build server.

m-mcgowan, 2015-12-23:
I understand a fix for this has been rolled out now to the compiler service.

closed 2015-12-23 by m-mcgowan