particle-iot/device-os #771: Disable double-tap of setup for Photon

monkbroc, 2015-12-16:
In the latest develop, double-tapping the setup button puts a Photon to sleep. Only a reset will bring it back up.

This should not be enabled by default in 0.4.8 as it is a user-visible change in behavior from previous versions.

Perhaps implement a `POWEROFF_FEATURE(ENABLED/DISABLED)` macro with it enabled by default for the Electron and disabled for the Photon?

[Firmware reference](https://github.com/spark/firmware/blob/2c52f307a90103234a99648e0e36d50918909650/system/src/main.cpp#L88)

technobly, 2015-12-16:
Agreed!  Thanks for the note on this.  We currently have a fix in place here: https://github.com/spark/firmware/pull/767

In the future we will likely a user option to enable single and double taps for the setup button.
* single = RSSI UX
* double = System Deep Sleep (aka Soft Power Down)

monkbroc, 2015-12-16:
Great.

As a side note, I like that #767 uses a capability flag `Wiring_SetupButtonUX` instead of a platform flag `PLATFORM_ID == 10`. It's way better in the long run to test for features instead of platform (like how browser detection moved to feature detection with Modernizr). This shouldn't be reserved for Wiring capabilities though. I think there should be a `hal/src/XXX/capabilities_impl.h` (or maybe a makefile) that defines all the capabilities defined for each platform like serial flash, availability of wifi, etc and that most of the `#if PLATFORM_ID == XX` should be removed from the firmware.

m-mcgowan, 2015-12-16:
sure, we have hal_platform.h for similar platform features. Totally agree that the code should focus on features being enabled rather than platform.

m-mcgowan, 2016-01-19:
Power off disabled on the photon by the PR listed above.

closed 2016-01-19 by m-mcgowan