particle-iot/device-os #776: USART ISR constantly entered due to Serial Overrun Error not handled correctly

jakeypoo, 2015-12-18:
Found a bug where P1/photon can seem like it's frozen due to the HAL_USART_Handler constantly going off when the ORE flag is set in USART_SR and the RXNE bit is not set.

ORE  = Overrun error flag in the USART Status Reg
RXNE = Rx buffer not empty flag
RXNEIE = ISR enable on RXNE flag set

Currently HAL_USART Handler doesn't handle the case where RXNEIE=1, RXNE=0, ORE=1 and the USART ISR goes off constantly. 

The UART ISR will go off if ORE is set and RXNEIE is enabled. Overrun errors are handled just fine most of the time since the RXNE bit is set on most occurrences and the ISR currently will read the USART DR if RXNE is set. The ORE bit is cleared by a read to USART_SR followed by a read to the USART_DR. **However, it's possible the ORE can be set as the RXNE is cleared.**

From the STM32F2xx reference manual 
   >This case can occur when the last valid data is read in
the RDR at the same time as the new (and lost) data is received. It may also occur
when the new data is received during the reading sequence (between the USART_SR
register read access and the USART_DR read access).

technobly, 2015-12-18:
@jakeypoo thank you for bringing this up!  It doesn't sound like there is an easy way to re-create this failure, you just have to wait for it?  Is there anything you can suggest to ensure you will see it occur?

Also if you have any suggestions on what exactly the fix is, that would be appreciated.  I had previously added some Overrun flag detection, Rx buffer flush and Overrun IT pending clearing to the handler but had commented it out for another day when I could put more thought into the need for it.

https://github.com/spark/firmware/blob/develop/hal/src/stm32f2xx/usart_hal.c#L387-L392

jakeypoo, 2015-12-18:
@technobly Yup I saw your commented out block of code. I'm going to put in a solution in the HAL_USART_Handler as this is a critical issue for me. I'll update here when I have a fix. Just wanted to create an issue since it might be a hard bug for others to find, and anyone using Serial1 is bound to encounter this. 

As for recreating the issue, see the snippet below for a somewhat complicated way to force the bug to occur. Run this code, press '1' in the USB serial , spam the Serial1 pins with some UART characters, press '2' in the USB serial. 

```c 
/*
 *  Test stub -- Trying to recreate an issue with the P1 crashing due to
 *               UART1 Overrun error
 *  Notes :
 *      Issue revolves around 
 *
 */

#include "application.h"

SYSTEM_MODE(SEMI_AUTOMATIC); 

void setup() {
    Serial.begin( 9600 ) ;
    Serial1.begin( 115200 ) ;
}

void loop() {
    char r ; 

    if( Serial.available() > 0 ){
        r = Serial.read() ;
        
        switch( r ){
            case '1' :      /* Step 1:  Stop the RX ISR then spam the UART with chars */
                *( __IO uint32_t * )( 0x4001100C ) &= ( ~0x20 ) ;   // USART1 CR1 @ x4001100C 
                Serial.println("*RX ISR disabled*"); 
                break ;
            case '2' :      /* Step 2 : Restart the ISR, should see the ORE flag set now */
                Serial.println("*RX ISR enabled*"); 
                if( *( __IO uint32_t * )( 0x40011000 ) & 0x8 ){
                    *( __IO uint32_t * )( 0x40011000 ) &= ( ~0x20 ) ;   // clear the RXNE flag, leave ORE set
                    Serial.println( "ORE Flag detected" ) ;             // this condition causes the bug 
                }
                *( __IO uint32_t * )( 0x4001100C ) |= 0x20 ; 
                break ;
            case '3' :                                                                                                                                                                                                                                                         
                Serial.println("");
                Serial.print(" UART1 SR : ");  
                Serial.println( *( __IO uint32_t * )( 0x40011000 ) );   // USART1 SR @ x40011000
                break ;
        }
        while( Serial.available() > 0 ){
            ( void )Serial.read() ;
        }
    }

    delay( 50 ) ;
}
```

jakeypoo, 2015-12-18:
Just wanted to confirm that this modification to HAL_USART_Handler fixes this issue : 
``` c
static void HAL_USART_Handler(HAL_USART_Serial serial)                                         
{
    if(USART_GetITStatus(usartMap[serial]->usart_peripheral, USART_IT_RXNE) != RESET)          
    {
        // Read byte from the receive data register                                            
        unsigned char c = USART_ReceiveData(usartMap[serial]->usart_peripheral);               
        store_char(c, usartMap[serial]->usart_rx_buffer);                                      
    }                                                                                          

    if(USART_GetITStatus(usartMap[serial]->usart_peripheral, USART_IT_TXE) != RESET)           
    {
        // Write byte to the transmit data register                                            
        if (usartMap[serial]->usart_tx_buffer->head == usartMap[serial]->usart_tx_buffer->tail)
        {                                                                                      
            // Buffer empty, so disable the USART Transmit interrupt                           
            USART_ITConfig(usartMap[serial]->usart_peripheral, USART_IT_TXE, DISABLE);         
        }
        else
        {                                                                                      
            // There is more data in the output buffer. Send the next byte                     
            USART_SendData(usartMap[serial]->usart_peripheral, usartMap[serial]->usart_tx_buffer->buffer[usartMap[serial]->usart_tx_buffer->tail++]);
            usartMap[serial]->usart_tx_buffer->tail %= SERIAL_BUFFER_SIZE;                     
        }
    }                                                                                          

    // If Overrun flag is still set, clear it
    if (USART_GetFlagStatus(usartMap[serial]->usart_peripheral, USART_FLAG_ORE) != RESET)      
    {
        (void)USART_ReceiveData(usartMap[serial]->usart_peripheral);
    }                                                                                                                                                                                           
} 
```

m-mcgowan, 2016-01-04:
@jakeypoo  Would you mind submitting a PR with the necessary changes?

jakeypoo, 2016-01-05:
Here you go: https://github.com/spark/firmware/pull/796 
Hope I did that correctly.

closed 2016-01-07 by technobly