particle-iot/device-os #784: System doesn't restore RGB color after RSSI UX

monkbroc, 2015-12-24:
When pressing the SETUP button once for the RSSI blink UX, [system_handle_single_click](https://github.com/spark/firmware/blob/develop/system/src/main.cpp#L162) takes over than releases control of the RGB LED. If the user application had taken control of the RGB LED, the previous color and control is not restored.

Control could be restored by doing `bool previousControl = RGB.controlled()` but there's currently no API to get the previous color.

Another argument to implement better partitioning of the RGB LED #569.

technobly, 2015-12-25:
I thought I noticed this as well so thanks for confirming @monkbroc I like the idea of simply sampling what the current color is going into this routine, and restoring when finished.  That would obviously be a quick throwaway fix pending the new RGB LED partitioning.

m-mcgowan, 2016-01-04:
I also noticed a delay when pressing the button and the LED changing - this leads so some user confusion about whether the button press has been registered.  I believe (but haven't proven) the delay was due to the time it takes to get the RSSI value from the modem - the LED was only changed after the RSSI was available. The new behavior is to change the LED as soon as the button press is registered - before getting the RSSI value, and then the LED bars are signalled once the RSSI value is known.

technobly, 2016-01-07:
Fixed with PR #792

closed 2016-01-07 by technobly