particle-iot/device-os #786: Define flag for interrupt context

monkbroc, 2015-12-25:
Several user issues stem from calling blocking functions from interrupt context.
http://community.particle.io/t/documentation-can-be-improved-on-interrupts/18528

Since the system firmware controls all interrupts, I suggest adding an interrupt context counter and turning some blocking operations like `delay` into no-ops inside interrupts. Maybe also adding a blinking RGB LED notification meaning blocking operation inside interrupt.

m-mcgowan, 2015-12-25:
I agree. I've long planned to have the aystem actively manage registered interrupt handlers and interrupt context. #257.  Having key blocking functions not block when invoked from an interrupt context is a great application of that. 

Although since this is a programming error, first and foremost we should document what is safe to call from an interrupt context and what isn't. The 2nd step is enforcement of that.  I feel a no-op might be too easily overlooked depending upon how the functionality is used. I'd rather have a new SOS code - for blocking function called from an interrupt. That way the user is unambiguously informed of their mistake.

m-mcgowan, 2016-01-22:
In this commit, I defined a isISR() function that's used to determine if the call is in an interrupt context. https://github.com/spark/firmware/commit/bc7d59b65bb8e22714b8d0186c5023747b91a10f

The HAL can add this check to any functions that are not safe to call from an ISR.

m-mcgowan, 2016-03-18:
Here's some initial thoughts on APIs that could have conditional behavior if HAL_isISR() returns true:

- delay() becomes a basic busy-wait delay without doing any background processing. (Or we simply don't delay at all.)
- Serial[X].write() doesn't block (as if blockOnOverrun() is false)
- TCP/UDP network operations. This may be device-specific and so handled in the HAL.
- other blocking functions...

We could have a new PANIC code "invalid function called from ISR", and clearly document these. This should be reserved for cases where the device would catastrophically fail anyhow, so we are here just providing a more localized panic code.

avtolstoy, 2020-02-28:
Most of the things listed above are already implemented nowadays, in many other cases the documentation lists things to avoid.

closed 2020-02-28 by avtolstoy