particle-iot/device-os #798: Serial.availableForWrite function

m-mcgowan, 2016-01-06:
https://community.particle.io/t/serial-availableforwrite-function/18831/4

We can check if the TX buffer is full or not to allow applications to implement flow control.

Ibuprofen, 2016-01-06:
:+1: Thanks for filing this!

m-mcgowan, 2016-01-09:
Currently, the hardware serial devices corrupt data if fed too quickly. I feel the default should be to block until the buffer is emptied, so the application doesn't have to concern itself with how fast it writes data by default.  For most, having the correct data output is more important than writing it quickly and getting corrupted data.    For cases where performance is critical, a new method. `blockOnOverrun(false)` can be called to disable blocking, and revert to the current non-blocking behavior that overruns the buffer.

The USB serial device write function includes a delay as a simple form of flow control. Instead, it should check if the buffer would overrun and block, similarly to Serial1. One additional point is that the buffer is not emptied if no USB host is connected, so the method should block only when a host is connected. This can be detected by checking the established linerate.

PR #812 (2016-03-10): Serial.availableForWrite() and Serial.blockOnOverrun() implementation, USB Serial fixes for STM32F2
Added Serial.availableForWrite() method for hardware USARTs and USB Serial.
Added Serial.blockOnOverrun(bool) method to set blocking/non-blocking behavior on TX buffer overrun (default is blocking).

This PR also includes a number of fixes for USB Serial driver on STM32F2.

Closes #798
closed 2016-03-10 by m-mcgowan