particle-iot/device-os #803: Rework EEPROM emulation

m-mcgowan, 2016-01-11:
The EEPROM emulation layer is in need of some rework to meet these requirements:

- the code is not duplicated, but shared between Core and STM32F2 devices (at least as much as possible - device differences notwithstanding.)
- the code is testable, both on device and off device
- the emulated EEPROM supports atomic (all or nothing) writes of multiple bytes in the presence of power failures
- off-device tests validate the expected flash wear (number of erases.)
- off-device tests validate atomic writes and non-corruption of data
 

Some suggestions for how to achieve this:

- the code can be separated into hardware neutral and hardware specifics. Much of the emulation algorithm is hardware neutral. 
- off device tests using the hardware neutral code, provided with an emulated flash memory (see RAMFlashStorage in `services/inc/flash_storage.h`)
- atomic writes can be achieved using a sentinel - https://community.particle.io/t/what-is-the-fuel-gauge-eeprom-write-with-low-battery/15975/18?u=mdma

m-mcgowan, 2016-01-18:
@monkbroc is looking into this

monkbroc, 2016-03-22:
Implemented in #871

closed 2016-03-22 by monkbroc