particle-iot/device-os #814: SPI Slave Support in Firmware

harrisonhjones, 2016-01-15:
Posting here for a customer. Would like to add firmware support for using the Photon/Spark Core as a SPI slave. 

This [gist](https://gist.github.com/chrismeyersfsu/3317769) might help. It's written for the arduino. 

This [page](http://www.gammon.com.au/spi) has a bit more in-depth information. 

Feel free to discuss this feature here.

AronRubin, 2016-02-03:
https://github.com/btmcmahan/Teensy-3.0-SPI-Master---Slave

dseemakurty, 2016-02-18:
I would find this extremely helpful also!

m-mcgowan, 2016-03-18:
@avtolstoy  Please add tests to verify that SPI.transfer() with a `nullptr` callback does block until the transfer is complete.

avtolstoy, 2016-03-20:
#919 fixes SPI.transfer() behavior and adds necessary tests.

closed 2016-04-03 by technobly